FreeJoy-Team/FreeJoy #2: Analog input

mahdiyari, 2019-12-11:
Hi
Buttons are working but I couldn't get analog inputs work
even without connecting anything to A0 - A7, the input values are changing (some kind of noise)
I see values using windows controller configuration and a few joystick/gamepad programs
All analog inputs are moving even without a connection
is there anything I'm doing wrong?

vostrenkov, 2019-12-11:
If analog input is left floated it gonna show you noise values. Try to connect analog inputs to ground to values near 0 (minimum) or to 3.3V to see values near 4095 (maximum). 
For normal operation you may want to connect potentiometer (or hall sensor, or anything else) to your analog pin to generate analog signal of range 0V - 3.3V

mahdiyari, 2019-12-11:
I did same thing and input was not stable
even when connecting to ground it jumps around 0 (not exactly 0)
I think it may be because of computers noise on the usb port (or maybe nearby speakers?)
Thanks for your response

vostrenkov, 2019-12-12:
It is not simple to achieve stable 12-bit input at bluepill board. You can use external RC or RLC filter to reduce noise or you can use software filter option. 
Soft FIR filter can be enabled using FreeJoyConfigurator utility (see [wiki page](https://github.com/vostrenkov/FreeJoy/wiki/Configuarion-with-visual-tool))or in main.h file

vostrenkov, 2019-12-24:
I found a way to reduce input noise a lot even without using digital filtering. ADC sampling time should be configured to maximum value (239.5 cycles). 
This fix will be included to the next release, but you can build from /dev branch to try it now (or change it yourself in your fork)

closed 2020-03-31 by vostrenkov

teeshq, 2021-02-16:
> 
> 
> It is not eazy to achieve stable 12-bit input at bluepill board. You can use external RC or RLC filter to reduce noise or you can use software filter option.
> Soft FIR filter can be enabled using FreeJoyConfigurator utility (see [wiki page](https://github.com/vostrenkov/FreeJoy/wiki/Configuarion-with-visual-tool)) or in main.h file

Hi what option is use to soft filtering a noise ??

vostrenkov, 2021-02-16:
> Hi what option is use to soft filtering a noise ??

Hi, filter slider is used for each axis on axis tab of configurator

pipes80-collab, 2021-02-16:
I have noticed that increasing the digital filter you mentioned almost solves the problem but if it is increased too much it slows down the slide