FreeJoy-Team/FreeJoy #3: USB HIDName is incorrect

betgear, 2020-01-07:
Hi,

Retrieved HID name is not correct in the PC app.

[image]

The name appears in the APP as 'r during stacking' which is obviously a partial string from somewhere random in memory

[image]

Additionally I cannot see to save a config file to the STM32, I tried creating a debug build to get more information for you, but unfortunately the debug build will not fit in the microcontrollers memory.

vostrenkov, 2020-01-07:
Hello,
looks like the application don't write default config at first startup. The reason can be flash memory at that area is not empty during first run. Please try to do full chip erase and try to load code again.

betgear, 2020-01-07:
Hi,

Yes, that makes sense. It works now :)

Why not use a checksum on the config data, if the checksum does not match, write the default configuration data with correct checksum.

vostrenkov, 2020-01-07:
I didn't this could happen :)
I will add some kind of protection to future commits.

closed 2020-01-07 by vostrenkov