FreeJoy-Team/FreeJoy #5: 12 Pulse Mechanical Rotary Encoder

0xcore, 2020-01-08:
Hi!

Is it possible to use 12 pulse incremental rotary enconders like this ? 

https://octopart.com/ec12e1220401-alps-7471732

Thanks!

vostrenkov, 2020-01-08:
Yes, it is possible to connect up to 64 rotary encoders.
Rotary encoders works like increment/decrement buttons. Rotation CW generates pushing one button, rotation CCW generates pushing another button. Push duration is configurable.

0xcore, 2020-01-08:
I want to use an encoder that uses only 3 wires for 12 positions.

http://www.leobodnar.com/shop/index.php?main_page=product_info&cPath=98_100&products_id=201

vostrenkov, 2020-01-08:
This is not encoder, but rotary switch. I'm not sure how it works, maybe it gives encoder-compatible output but it can be not. 
I guess this switch is only suitable to Bodner's controllers.

mackyt, 2020-01-08:
I am guessing based on the photo but it looks like that bodnar "encoder" is a rotary switch with a PCB that contains a switched potential divider output. ie (with 5V/GND in) it gives an analog 0 to 5V out in 12 steps. So it could be used with freejoy as a normal analog input but I suspect that would not be very useful. (depends on game/application)

vostrenkov, 2020-01-08:
FreeJoy have AxisToButtons feature, making it possible to convert analog input to buttons. But now it supports up to 10 buttons per axis, so this switch won't work correctly (if it provides analog output)

0xcore, 2020-01-14:
It is to map 12 positions of the switch with keyboard. F1, F2, F3... Sim Racing steering wheel needs many inputs

[image]

vostrenkov, 2020-01-14:
Current version of FreeJoy does not support Bodnar's 3-wire switches and i'm not sure when i will implement it. But next release (upcoming in a week or so) will support shift registers 74HC165 and CD4021 for connection big amount of buttons (up to 128) with just 3 wires. 
You still can connect buttons in matrix style, for 12 inputs you need 12=3x4 matrix (3+4=7 pins)

0xcore, 2020-01-14:
Thanks! @vostrenkov 74HC165 or CD4021 is perfect, more inputs problem solved.

vostrenkov, 2020-01-14:
Also I will add 2 more buttons to AxisToAnalog feature to achieve 12 buttons per axis (now it is only 10). It may solve problem with Bodnar's switches

closed 2020-01-20 by 0xcore