FreeJoy-Team/FreeJoy #7: Missing Shift Register functionality

godor2008, 2020-01-18:
On main page there is screenshot that has "Shift registers" tab. But when I launched config utility - there is no such tab :( Are there any plans/intentions to add such functionality?

betgear, 2020-01-19:
This missing TAB is in the dev branch of the configurater code. I assume once the developer is happy with the changes and he has tested the changes to his satisfaction , the branches will be merged.

vostrenkov, 2020-01-19:
Yes, shift registers functionality is nearly ready to be merged and released

vostrenkov, 2020-01-21:
Now you can try shift registers functionality. Please check out [release page](https://github.com/vostrenkov/FreeJoy/releases) and [manuals](https://github.com/vostrenkov/FreeJoyConfigurator/wiki/Shift-registers).

godor2008, 2020-01-21:
Cool man!

godor2008, 2020-01-21:
Not issue, rather suggestion. I think it would be nice to be able to select GND or VCC at Shift register inputs.

vostrenkov, 2020-01-21:
It is only reasonable for CD4021. 74HC165 has both inverted and non-inverted output.
Maybe i will add this later, now you can invert buttons on buttons tab

godor2008, 2020-01-21:
Ah, got it, will try when my boards arrive

vostrenkov, 2020-02-20:
Now shift registers have pull-up and pull-down selection

closed 2020-02-20 by vostrenkov