FreeJoy-Team/FreeJoy #10: Can't read/write config in release 1.0.2

godor2008, 2020-02-02:
Using configurator from release v1.2.2
Using hex v1.02

When click Read config or write config, nothing happens
[image]

vostrenkov, 2020-02-02:
Try to do full chip erase and program it again. 
Please show me the log window from ST-Link v2 utility.

godor2008, 2020-02-02:
01:25:46 : ST-LINK SN : A37E050132124647524B4E00
01:25:46 : V2J29S7
01:25:46 : Connected via SWD.
01:25:46 : SWD Frequency = 4,0 MHz.
01:25:46 : Connection mode : Normal.
01:25:46 : Debug in Low Power mode enabled.
01:25:46 : Device ID:0x410 
01:25:46 : Device flash Size : 64KBytes
01:25:46 : Device family :STM32F10xx Medium-density
01:25:54 : Flash memory erased.
01:26:01 : [FreeJoy_v1_02b9.hex] opened successfully.
01:26:01 : [FreeJoy_v1_02b9.hex] checksum : 0x00B7F48A 
01:26:10 : Memory programmed in 4s and 687ms.
01:26:10 : Verification...OK
01:26:10 : Programmed memory Checksum: 0x00B7F48A

vostrenkov, 2020-02-02:
I just tried to do clean installation of configurator and flashed firmware 1.02 from release to 2 different boards with 64 and 128kb memory. I can say it works quite fast and stable.
So i guess this problem is host computer's system or/and hardware dependent. Which version of Windows do you use?
How long is the USB cable connecting your board to the PC?
Do you have this cable connected to USB hub or directly to USB port?

godor2008, 2020-02-02:
Windows 10

> How long is the USB cable connecting your board to the PC?
Instantly

>Do you have this cable connected to USB hub or directly to USB port?
Directly

I used same board and flashed your previous release just for test. That time it worked fine

godor2008, 2020-02-02:
Just tried previous release, and now get same issue. Data not writing. May be it is something with board

vostrenkov, 2020-02-02:
Rebooting your PC may help

godor2008, 2020-02-02:
It is detected by Win 10
[image]

godor2008, 2020-02-02:
Reboot helped. I downloaded latest release. When installing it, new driver is also installed. And seems this driver is not active until reboot

vostrenkov, 2020-02-03:
It is quite strange behaviour. Maybe problem is in HID library used in configurator app. I already had problems with hanging driver by this library but I can't catch this because it happens really seldom

godor2008, 2020-02-03:
Windows style solution -> reboot your PC seems to be really actual in this case for that driver :)

closed 2020-06-16 by vostrenkov