FreeJoy-Team/FreeJoy #11: All axes are displayed even if no axe is defined

godor2008, 2020-02-05:
[image]
I selected only buttons in latest configurator. But Windows game controller setup shows that there are lot of axes

vostrenkov, 2020-02-05:
Yes, it is ok.
USB HID descriptor is the same for all configurations. You will see 8 axes and 128 buttons even if nothing selected. Not selected axes always have 0 value and not selected buttons always have false state so they won't bother you to much

godor2008, 2020-02-05:
And will it be implementable in future to remove not used axis/buttons?

vostrenkov, 2020-02-05:
It is possible, but i don't see a lot of reasons to do this now. I will add this task to my list with low priority

vostrenkov, 2020-02-23:
Dynamic HID config option is now available in version v1.3.4. Check out releases page

closed 2020-03-09 by vostrenkov