FreeJoy-Team/FreeJoy #13: Toolchain used?

franukis, 2020-02-20:
Hi guys,

I'd been playing with the sources, to become familiar with and potentially contribute something in the future. Problem is I am not that familiar with STM32 and last time I used them it was with a company-paid Keil IDE. 

So, I downloaded and installed the new STM32CubeIDE, but it wants to convert the project to a different format, and tells something about it being an MX format used. My question is: which toolchain (IDE,compiler...) are you using? I assume using the same one would be the most straightforward approach, and I have no preferences, honestly.

When I convert the project to the STM32CubeIDE as a STM32 project, I can't build it, and cannot even modify the project properties, like if I'd like to compile for a different uC of the same family, I cannot choose the device. If I convert it to MX project, then most of the source is replaced by auto generated code, and even the main.c is completely different (placeholder like)...

Could you please help me here?

Thanks!!

vostrenkov, 2020-02-20:
Hello,
FreeJoy is a Keil uVision v.5.24 project and you can use it with MDK ARM compiler, but it still can be built with armgcc and makefile if you are using Linux or you just like this way of building projects. Code size does not exceed 32K (and will not in the near future), so free licence of Keil uVision can be used. 
There are two targets in the project: Application and Bootloader. Both of them can be built and flashed with Keil while makefile does not have target for building bootloader for now. Bootloader makes it possible to load .bin files of Application from configurator utility and isn't necessary if you are always flashing code with ST Link programmer.  

For more information how to build FreeJoy from sources take a look at wiki page

franukis, 2020-02-21:
Thanks!!! I‘ll start playing with Keil, then!

closed 2020-02-21 by franukis