FreeJoy-Team/FreeJoy #15: DPad configuration

ScheintodX, 2020-02-25:
Hi. Very impressed from this project and it's configuration tool.

What I don't seem to find is a option to use two *digital* inputs for one axis.

(What I'm trying to do is to usbify my old Competition Pro 2 axis 2 buttons (or any old arcade joystick for that matter.) https://upload.wikimedia.org/wikipedia/commons/thumb/9/96/CompetitionPro_2nd_Series_Inside.jpg/1600px-CompetitionPro_2nd_Series_Inside.jpg)

I did find the 4 pov/hat controls but they end up being analog axis 8+ and I'd like mine to be axis 0/1.

vostrenkov, 2020-02-25:
You can try with axis from buttons feature. Create two buttons and set them as source of analog axis. then set step to maximum and you will get 0-1 output at this axis

vostrenkov, 2020-02-25:
[image]
[image]

If i understand you correctly this is the solution for you. Pressing button 1 sets axis to minimum, pressing button 2 sets axis to maximum. I use radiobuttons to have button 1 pressed at startup (in case of normal buttons axis is set to center at startup).

ScheintodX, 2020-02-25:
Hi. Thanks for your quick reply. But this doesn't centers the axis if no button is pressed.

vostrenkov, 2020-02-25:
I afraid you need additional button for center position to make it possible without code modification. This feature is mostly made for using rotary encoder as axis source and it shouldn't return to center after releasing buttons.

ScheintodX, 2020-02-25:
Ok. So at least I'm not too stupid to find it :)

I'm currently looking at the sources. (Thanks for the gcc makefile!). Three possibilities come to mind: 
* Add a config for dpad to change its axis (0/1 ...) / or add a couple more dpads with axis starting at 0
* Add a type similar to dpad ... lets say "digital axis" or so which feed analog values int axis
* add a setting to axis to not increment/decrement but set max values or zero if not pressed.

Which one would you say makes sense?

vostrenkov, 2020-02-26:
Sorry for late answer.
The last is the easiest one in my opinion.
In file analog.c in function AxesProcess() find block about increment and decrement value and change logic a little for your needs

ScheintodX, 2020-03-10:
Thanks for the effort. I meanwhile got away by simply swapping the axis on the host with jstest-gtk and then moved on to the other part of the project where I connected a c64 matrix keyboard to another bluepill where in this case I had to write the whole firmware myself.
But I will try it out as soon as I take my dev tools out of the closet the next time.

Just one thought: If there was this option available in the config tool like lets say a checkbox for "return to zero" or something this project would be a real alternative for people currently buying stuff like this: https://www.ebay.de/itm/Zero-Delay-Arcade-USB-Encoder-PC-to-Joystick-MAME-5Pin-Sanwa-Drucktaster-AC424/133300896491?hash=item1f095a42eb:g:JG8AAOSwNAVeGbbv

vostrenkov, 2020-03-10:
I can not promise I will add this feature but it is possible in the future