FreeJoy-Team/FreeJoy #17: PWM output control by one of the AXIS

jieweiyang, 2020-02-28:
Is there any chance to have a timer pin for PWM output, and control by one of the axis?

vostrenkov, 2020-02-28:
I'm working at configurable PWM outputs for LEDs now but I supposed them to have static duty cycle during runtime (configurable in configurator). I can not promise you I will add duty cycle control by axis but it is possible

jieweiyang, 2020-02-28:
Static configure by tools will be good enough! Axis control will be better but not mandatory.  Appreciate for your great work!

jieweiyang, 2020-03-07:
PWM enabled, but 500Hz will have some flashing. Hopefully can bring up to the clock to 1KHz?

vostrenkov, 2020-03-07:
To be honest i have not tested PWM channels with LEDs. I'll make it better when i get my LEDs to me

vostrenkov, 2020-03-07:
I've found some LEDs and tested PWM with them. I'm not able to see any flashing and i'm sure i shouldn't. Have you real tests with blinking at 500Hz or it is your assumption?

jieweiyang, 2020-04-06:
My design is PWM to N-MOS to drive the LED. Found the jumper cable between the bulepil and the test board not stable and caused the flashing.  After program the freejoy in my existing PCBs, everything works fine now, no flashing anymore.