FreeJoy-Team/FreeJoy #18: MLX90393 16bit AD support

SarwarN88, 2020-03-02:
I just want to know if its possible to add this hall sensor that uses i2c communication to create joystick device using FreeJoy with STM32? There are i2c files in the drives folder (**stm32f1xx_hal_i2s.h**) and (**stm32f1xx_hal_i2s.c**) but I don't know how to use them. I have a code to run the mlx90393 with wire.h library using Arduino Leonardo and **Arduino Joystick Library by **MHeironimus**** to make the joystick but I want to use the power of STM32 with better clock speed. 
Thanks

vostrenkov, 2020-03-02:
FreeJoy does not support I2C sensors at this moment. 
But if you are not going to have it fully configurable by tool you can hardcode support of I2C sensors into the firmware. 
All stuff about analog channels is placed in analog.c and sensors.c files.

There is a real chance that I2C sensors support will be added to the project soon (after external ADCs)

SarwarN88, 2020-03-02:
Thanks for the quick replay. I think I will modify wire.h library for the STM32 and use Arduino Joystick Library for my Rudder pedals at the moment. I will be using FreeJoy for my Throttle joystick. In the form it said the axis support up to 16bit but in the configuration tool it said max 12bit resolution? I know it is related to the STM32 12bit ADC resolution but if external ADC support added will we be able to set the range to 16bit?

vostrenkov, 2020-03-02:
Internal ADC of STM32 has resolution of only 12bits. External sensors TLE501x have about 15.5 bits. So the code map raw data from ADC or sensor to 16bit value (from -32767 to 32767) to have same range of output values for any chosen axis source.
You can set resolution to lower level - output value will remain 16bits but effective resolution will be decreased

SarwarN88, 2020-03-02:
The problem with achieving 16bits resolution is its very sensitive to electronic noise. Its almost impossible to make 16bits noise free without not using advance shielding and smoothing function. Smoothing function is also limited for joystick that require fast interaction.  My aim is to achieve 13bits resolution with noise free by shielding all of my cables, sensors and ground all the float pins. with the 72 MHz clock speed of STM32 I wounder how long it will take to smooth sam of lets say 16 variable values to return 1 value? I need to check that once my STM32 arrives.

vostrenkov, 2020-03-02:
If you are going to use external ADC your sampling rate will be only limited by SPI/I2C performance of used ADC chip. STM32 can clock SPI with frequency up to 32MHz. 
Foe example for TLE501x 1.15MHz is used because of limitation of the chip

SarwarN88, 2020-03-04:
Thanks for that I just find out that the MLX90393 can run in both SPI and I2C communication. The I2C max frequency is 400KHz in the fast mode while the SPI mode can run at max 10MHz. Since I am using only one sensor its good option to go with the SPI mode.

vostrenkov, 2020-03-04:
I'm waiting my MLX90393 and ADS1115 from China to make some tests and add them to the project. So if you are not in a hurry just wait for that :)

SarwarN88, 2020-03-05:
That will be great. The MLX90393 has Arduino library that you can use for the testing. I am currently building mechanical parts for my flying Rudder pedals and it will probably going to take me another 3 months as I am working on it once a week. here is the picture of current state of the building.
   
[image]

vostrenkov, 2020-03-05:
Are you sure you really need 3d sensor for rudder? TLE5011/TLE5010 is a good choice for this application

SarwarN88, 2020-03-05:
I only use one axis from the sensor. I used MLX90393 because it was available option when I did the testing. For the toe brakes I will use A1324LUA-T with STM32 on board ADC.

vostrenkov, 2020-03-05:
A1324LUA-T has minimum supply voltage 4.5, be careful

SarwarN88, 2020-03-05:
I will use level shifter from 3.3V to 5V.

jieweiyang, 2020-03-07:
beside those ADC(A*) input pins, most of the pins of STM32F103C8T6(the bluepil) are 5v tolerance.

closed 2020-06-16 by vostrenkov