FreeJoy-Team/FreeJoy #21: STM Cube IDE 1.2.0 does not run source code

tiagodsnts, 2020-03-12:
Hi everyone,

STM Cube IDE 1.2.0 does not run the source code. There are several errors
Could someone help me how I can make modifications by the source code, should I use Cube IDE or CubeMX?
On the stm.com it is difficult to find old versions of CubeMX. Thanks.

vostrenkov, 2020-03-13:
This is not Cube IDE project. Cube MX was only used for hardware configuration. Use Keil uVision instead or build with makefile

tiagodsnts, 2020-03-18:
Thanks. Now i install Keil uVision 5.29. But, when i erase chip and select download show "Error: Flash download failed - Could not load file "FreeJoy\FreJoy.axf""

vostrenkov, 2020-03-18:
What do you have in log window after compilation?

tiagodsnts, 2020-03-18:
Flash > Erase 
Flash > Download 

Full Chip Erase Done.
Flash Erase finished 16:45:53
Load "FreeJoy\\FreeJoy.axf" 
_____^
*** error 56: cannot open file
Error: Flash Download failed  -  Could not load file 'FreeJoy\FreeJoy.axf'
Flash Load finished at 16:46:24

vostrenkov, 2020-03-18:
Do you have the code compiled successfully?

tiagodsnts, 2020-03-18:
No.
Debug > Start stop debug session

Load "FreeJoy\\FreeJoy.axf" 
_____^
*** error 56: cannot open file
Error: Flash Download failed  -  Could not load file 'FreeJoy\FreeJoy.axf'
Flash Load finished at 17:08:24

Projetc > Build Target

Build started: Project: FreeJoy
*** Using Compiler 'V5.06 update 6 (build 750)', folder: 'C:\Keil_v5\ARM\ARMCC\Bin'
Build target 'FreeJoy'
compiling config.c...
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(51): warning:  #47-D: incompatible redefinition of macro "__MPU_PRESENT"  (declared at line 158 of "..\Drivers\CMSIS\CM3\DeviceSupport\ST\STM32F10x\stm32f10x.h")
   #define __MPU_PRESENT             0U       /*!< Other STM32 devices does not provide an MPU  */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(52): warning:  #47-D: incompatible redefinition of macro "__NVIC_PRIO_BITS"  (declared at line 160 of "..\Drivers\CMSIS\CM3\DeviceSupport\ST\STM32F10x\stm32f10x.h")
  #define __NVIC_PRIO_BITS           4U       /*!< STM32 uses 4 Bits for the Priority Levels    */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(53): warning:  #47-D: incompatible redefinition of macro "__Vendor_SysTickConfig"  (declared at line 161 of "..\Drivers\CMSIS\CM3\DeviceSupport\ST\STM32F10x\stm32f10x.h")
  #define __Vendor_SysTickConfig     0U       /*!< Set to 1 if different SysTick Config is used */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(72): error:  #101: "NonMaskableInt_IRQn" has already been declared in the current scope
    NonMaskableInt_IRQn         = -14,    /*!< 2 Non Maskable Interrupt                             */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(74): error:  #101: "MemoryManagement_IRQn" has already been declared in the current scope
    MemoryManagement_IRQn       = -12,    /*!< 4 Cortex-M3 Memory Management Interrupt              */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(75): error:  #101: "BusFault_IRQn" has already been declared in the current scope
    BusFault_IRQn               = -11,    /*!< 5 Cortex-M3 Bus Fault Interrupt                      */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(76): error:  #101: "UsageFault_IRQn" has already been declared in the current scope
    UsageFault_IRQn             = -10,    /*!< 6 Cortex-M3 Usage Fault Interrupt                    */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(77): error:  #101: "SVCall_IRQn" has already been declared in the current scope
    SVCall_IRQn                 = -5,     /*!< 11 Cortex-M3 SV Call Interrupt                       */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(78): error:  #101: "DebugMonitor_IRQn" has already been declared in the current scope
    DebugMonitor_IRQn           = -4,     /*!< 12 Cortex-M3 Debug Monitor Interrupt                 */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(79): error:  #101: "PendSV_IRQn" has already been declared in the current scope
    PendSV_IRQn                 = -2,     /*!< 14 Cortex-M3 Pend SV Interrupt                       */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(80): error:  #101: "SysTick_IRQn" has already been declared in the current scope
    SysTick_IRQn                = -1,     /*!< 15 Cortex-M3 System Tick Interrupt                   */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(83): error:  #101: "WWDG_IRQn" has already been declared in the current scope
    WWDG_IRQn                   = 0,      /*!< Window WatchDog Interrupt                            */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(84): error:  #101: "PVD_IRQn" has already been declared in the current scope
    PVD_IRQn                    = 1,      /*!< PVD through EXTI Line detection Interrupt            */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(85): error:  #101: "TAMPER_IRQn" has already been declared in the current scope
    TAMPER_IRQn                 = 2,      /*!< Tamper Interrupt                                     */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(86): error:  #101: "RTC_IRQn" has already been declared in the current scope
    RTC_IRQn                    = 3,      /*!< RTC global Interrupt                                 */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(87): error:  #101: "FLASH_IRQn" has already been declared in the current scope
    FLASH_IRQn                  = 4,      /*!< FLASH global Interrupt                               */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(88): error:  #101: "RCC_IRQn" has already been declared in the current scope
    RCC_IRQn                    = 5,      /*!< RCC global Interrupt                                 */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(89): error:  #101: "EXTI0_IRQn" has already been declared in the current scope
    EXTI0_IRQn                  = 6,      /*!< EXTI Line0 Interrupt                                 */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(90): error:  #101: "EXTI1_IRQn" has already been declared in the current scope
    EXTI1_IRQn                  = 7,      /*!< EXTI Line1 Interrupt                                 */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(91): error:  #101: "EXTI2_IRQn" has already been declared in the current scope
    EXTI2_IRQn                  = 8,      /*!< EXTI Line2 Interrupt                                 */
[... cut ...]
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(103): error:  #101: "USB_LP_CAN1_RX0_IRQn" has already been declared in the current scope
    USB_LP_CAN1_RX0_IRQn        = 20,     /*!< USB Device Low Priority or CAN1 RX0 Interrupts       */
../Drivers/CMSIS/Device/ST/STM32F1xx/Include/stm32f103xb.h(104): error:  #101: "CAN1_RX1_IRQn" has already been declared in the current scope
    CAN1_RX1_IRQn               = 21,     /*!< CAN1 RX1 Interrupt                                   */
..\Src\usb_hw.c: 3 warnings, 30 errors
"FreeJoy\FreeJoy.axf" - 360 Error(s), 36 Warning(s).
Target not created.
Build Time Elapsed:  00:00:04

vostrenkov, 2020-03-18:
I just made a clean clone of the repo and everything is ok. 
Please delete your project folder and make a new clone of the repository. Then go to YourRepoFolder/MDK-ARM folder and open FreeJoy.uvprojx filein Keil uVision. 
When the project is loaded press F7 or "Build" button. Project should be built successfully after that

tiagodsnts, 2020-03-18:
It worked a lot
Thanks for your help
I'm learning to play with the bluepill board

closed 2020-03-20 by vostrenkov