FreeJoy-Team/FreeJoy #22: IS THERE ANYTING WRONG WITH ENCODER?

hde610, 2020-03-17:
When I choose  physical NO. as encoder A and encoderB,all of the physical icons begine to light green quickly。  And it cannot write to the device when any physical NO. was setted as encoder mode. The physical NO. mode automatically changes to normal button mode when reads config from device after Fjoy is restarted。

encoder：  EC11
with 74hc165D，5v pull up。

vostrenkov, 2020-03-17:
Please save the configuration to file and upload somewhere i could download it

vostrenkov, 2020-03-17:
Do you have your 74HC165 connected to 5V power line? If yes please use 3.3V instead and 3.3V pull up then

hde610, 2020-03-21:
> Do you have your 74HC165 connected to 5V power line? If yes please use 3.3V instead and 3.3V pull up then

Please tell me your email address，that I could email the documents abuout setings to you，3Q.

vostrenkov, 2020-03-21:
yuvostrenkov@gmail.com

hde610, 2020-03-28:
what s the problem？ Did u sove it？

vostrenkov, 2020-03-28:
I solved problem about reading config from device, but haven't tried encoders physically yet

hde610, 2020-03-28:
Is this a problem that  CS and DATA(miso) pins have to be adjacent，but I  separated them？

hde610, 2020-03-28:
Is there any problem in pulling up with 10K resistor

vostrenkov, 2020-03-28:
10K pull up register is okay. About connection refer to manual: https://github.com/vostrenkov/FreeJoyConfigurator/wiki/Shift-registers

vostrenkov, 2020-03-31:
I have checked your config with my 74HC165 registers (10 KOhm pull-up) and encoder. Everything works fine.
Can you show your connections schematics?

hde610, 2020-04-03:
After upgrading FreeJoyConfigurator to the latest version, the problem of reading and writing has been solved. &nbsp;It Can be recorded as an encoder.

After only the encoder is set with a higher button debounce time value（50ms）, the physical button flashes green in the&nbsp;left&nbsp;, but the logical button is not triggered in the&nbsp;right&nbsp;, which should be no problem. &nbsp;However, when both encoder and toggle switch are provided, the toggle switch will be triggered&nbsp;&nbsp;flashing. &nbsp;In other words, It seems that the toggle switch and encoder cannot exist at the same time.

The above experiment was carried out with 3.3v pull-up，using 74hc165d. &nbsp;B3-SCK；B5-CS;B6-DATA。

vostrenkov, 2020-04-03:
Next update it will be possible to set button delay time individually for each button. This should help with your problem

closed 2020-06-16 by vostrenkov