FreeJoy-Team/FreeJoy #24: Request: Enumerate as a keyboard using PWM for analog inputs

DeadlyDad, 2020-03-22:
Years back, I wrote a GlovePIE script to convert analog inputs from my steering wheel into PWM signals that emulated madly hammering on keyboard keys.  Surprisingly, the PWM keyboard mode was absolutely _indistinguishable_ from native steering wheel mode in the games I tried and even worked **flawlessly** in things like Flash games.  Is that doable with this library?  It would open up a **LOT** of options.

vostrenkov, 2020-03-23:
This is interesting option, it may be done but not in near future. First of all we need support for devices such as new analog sensors and external ADCs

DeadlyDad, 2020-03-29:
Okay, but please give **serious** consideration to giving it a higher priority, as it would allow adding analog control to games and apps that only support the keyboard.  (For example, games that use WASD for movement would *finally* be able to be fully analog--with a foot tilt board, for instance.)