FreeJoy-Team/FreeJoy #25: Hi, just can't find any installer

freakerman21, 2020-04-04:
I cannot find any way of installation.

vostrenkov, 2020-04-04:
[image]

https://github.com/FreeJoy-Team/FreeJoy/releases
https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases
https://github.com/FreeJoy-Team/FreeJoyConfigurator/wiki
https://github.com/FreeJoy-Team/FreeJoy/wiki

freakerman21, 2020-04-04:
So I download it and have two files. One .bin and one .hex. What should I do next??
Sorry I don't know how to do this.

vostrenkov, 2020-04-04:
You need to program your BluePill board with .hex file. There are instructions at wiki page: https://github.com/FreeJoy-Team/FreeJoy/wiki/Quickstart-guide

After programming you should install configurator utility and make needed setup for your application. Instructions are also available at wiki:
https://github.com/FreeJoy-Team/FreeJoyConfigurator/wiki

freakerman21, 2020-04-04:
Yes, I have programmed my blue pill but I don't see any installer or .exe or anything like that in any folder zip. 
I have also the OSH Studio and it works but I don't know how to make potenetiometers to work.
I am trying to make an analog clutch but I can't.

freakerman21, 2020-04-04:
[image]

vostrenkov, 2020-04-04:
Use links from my first reply to get zip with installer of FreeJoy Configurator.

closed 2020-06-16 by vostrenkov