FreeJoy-Team/FreeJoy #34: 4-directional Stick Switch + Encoder + Center-push button

0xcore, 2020-06-29:
Hi,

Does anyone know if it is possible to use with Freejoy ?

[image]
[image]

https://tech.alpsalpine.com/prod/e/html/multicontrol/switch/rkjxt/rkjxt1f42001.html

I have seen that it is possible with Arduino & Teensy but I suppose it would need some modification to work with FreeJoy.

Arduino
https://github.com/darjusp/test

Teensy
https://www.xsimulator.net/community/threads/teensy-lc-arduino-scripts.9681

Thanks!

vostrenkov, 2020-06-29:
Hi,
I don't see a reason it should not work here. You dont even need external pulling resistors, bluepill provides internal pull-ups and pull-downs. Just connect terminals as ordinary buttons (Button VCC or Button GND) and set encoder inputs as Encoder A and B, buttons inputs as Button Normal or whatever you need

jieweiyang, 2020-06-30:
It work as 5 buttons + 1 encoder, nothing special.

0xcore, 2020-07-01:
I'm going to try it.

Thanks!

0xcore, 2020-07-16:
Hi,

I don't know what happens but when I press the 4 directions stick in any of the four directions (A, B, C, D) the central push button (5) is also pressed. Pressing only the central push button (5) works as expected.

I have made the following connections:

[image]

I have made the following connections:

A, B, C, D -> Button_VCC - POV Up, Down, Right, Left
5 -> Button_VCC - Normal Button
7,8 -> Button_VCC - Encoder A, Encoder B
6,9 -> +5V
10 -> GND

[image]

vostrenkov, 2020-07-17:
This is exactly what logic table from datasheet says:
[image]

Sorry i didnt mentioned this earlier. I will add POV_Center option for buttons settings for handling this state

vostrenkov, 2020-08-04:
Please check v1.6.2. POV_Center added to buttons options for support this hardware

0xfgarcia, 2020-08-04:
Work like a charm. Good Job!

closed 2020-08-04 by 0xcore

memob777, 2020-08-29:
Is possible for somebody to share the legs of the RKJXT1F42001 and how to connect them plz
I cant find the data on the data sheet
[image]

AndyPeacock, 2020-11-24:
Hi, I'm about to try and use the same switch and its great to hear that it should work. I have some quick questions about connections though, hopefully someone can help. I'm planning to use a 4021 shift register to connect this up as I have done for all other connections. So far if I understand it correctly I have been connecting my switches between input to the shift register and ground (using [these ](https://oshpark.com/shared_projects/ta3Mpmcz) boards). So am I right in thinking that A,B,C,D, EncoderA, EncoderB and Com would all be connected to their own pin on the 4021, and Push and Encoder Com would go to one of the ground connections on the 2x8 connector? 

Sorry if this isn't the right place to post this.

Thanks

vostrenkov, 2020-11-25:
Right, both Com pins should be connected to system ground and signal pins to shift register's input

geologic, 2020-12-09:
Sorry to hijack this post, but how about connecting 2 of these switch? At "Button Config" tab i can see Pov1_ to Pov4_, but only one Encoder_A and Encoder_B

AndyPeacock, 2020-12-09:
I think the encoders work differently. If you just set one of your encoder hardware pins to EncoderA and the other to EncoderB for each of your switches I think that will work. I don't have my switches hooked up yet, but this is how it works for standard rotary encoders for me.

geologic, 2020-12-10:
Thanks @AndyPeacock, i'll try that