FreeJoy-Team/FreeJoy #36: Force Feedback feature

SeriouSerg, 2020-07-09:
Is it possible?

vostrenkov, 2020-07-09:
It is planned for furure releases

ProFiLeR4100, 2020-09-14:
Will it support STM32F103 board or it will work only on STM32F4 (seen in another issue)?

vostrenkov, 2020-09-14:
Not sure. Maybe I will manage to fit some kind of lite ffb version to F103

SeriouSerg, 2020-09-18:
Yury, can we talk about FFB feature? How to contact you?

vostrenkov, 2020-09-18:
Discord Magners#1207

ahuxtable, 2020-12-19:
if you want ffb, take a look at the emc project, its not free or open but for $10 you get a full wheel/pedals controller software for stm32f103 or f407 with button matrix etc. Between that and freejoy you can have a great sim setup. i have run mmos and emc and emc wins hands down. Its still in active development too.