FreeJoy-Team/FreeJoy #37: Multiple Encoder Connection

0xcore, 2020-07-21:
I am connecting several encoder in chain with 1N4148 diodes but I can't get it to work. The encoders do nothing when I connect the diodes according to the attached scheme, without diodes it works but causing multiple unwanted pulses between the two encoders.

PEC11R Encoders:
http://www.farnell.com/datasheets/2360546.pdf?_ga=2.13033214.817435842.1595327489-134403827.1586878496

Freejoy Config:
B12,B13 as Column Button
B14,B15 Column Row

Can FreeJoy work with this setup?

[image]

[image]

[image]

vostrenkov, 2020-07-21:
It should work, but there may be a mistake in manual, so swap row and column assignments. I gonna check it also a bit later

0xcore, 2020-07-21:
Enconders work when swapping row and column but causing multiple unwanted pulses between the two encoders.

vostrenkov, 2020-07-22:
I found a bug with encoder non-atomic access to buttons reading, wait for fix soon

0xcore, 2020-07-23:
Thanks Yuri, I see another problem regarding encoders, maybe it's related.

[image]

[image]

vostrenkov, 2020-07-23:
Thanks for the report. It is only visual issue and should not impact functionality. It will be fixed also

vostrenkov, 2020-08-04:
Please check v1.6.2. It should fix the issue

0xcore, 2020-08-05:
Fixed, thanks!

Tested with 2 encoders.
B3,B4 as row. Encoder A & B pins
B5,B6 as column. Encoder Com

[image]

[image]
[image]
[image]

closed 2020-08-05 by 0xcore