FreeJoy-Team/FreeJoy #38: Analog Axes to Buttons

0xfgarcia, 2020-08-03:
Hi,

I can't get axes to buttons conversion to work. I'm trying with a 12 position rotary switch with a voltage divider but only 8 positions work. Analog raw read at positions 1 to 4 remains at 32.767, 5 to 12 between 28.0xx to -24.0xx

Is this setting correct in Freejoy Configurator?

Thanks!

1-4 Positions
[image]

5 Position
[image]

12 Position
[image]

Axes to buttons
[image]

vostrenkov, 2020-08-03:
Hello, 
do you connect your divider to 5V or to 3.3V?
If you show your connections it would be easier to analyze the issue

0xfgarcia, 2020-08-03:
I have connected it to 5V, GND, A0, 1K resistors between pins. I don't know how to test in STM32 :( I have made a test in Arduino, analog reading seems correct, different output for each 12 positions of the rotary switch, values 90 to 1023.

vostrenkov, 2020-08-03:
STM32 have 3.3V supply voltage, so maximum input voltage should not exceed 3.3V. 
Connect your divider to 3.3V and it will show you something similar to what you got with arduino

0xfgarcia, 2020-08-04:
Connected to 3V, now Axes Config works as expected. Thanks Yuri!

closed 2020-08-04 by 0xfgarcia