FreeJoy-Team/FreeJoy #40: Support for F4

Avalonnw, 2020-08-19:
Is there gonna be a support for F4 boards like this [one ](https://www.aliexpress.com/item/4000346695587.html?spm=a2g0o.productlist.0.0.62c239a2iDjmyw&algo_pvid=51cf35bd-2160-4b53-937f-203bfd7f2d0d&algo_expid=51cf35bd-2160-4b53-937f-203bfd7f2d0d-0&btsid=0ab6f8ad15978121533632586e453a&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)?

vostrenkov, 2020-08-19:
Possibly yes but more like as a side project

bayerla, 2020-08-19:
What will be better with a F40x ?

vostrenkov, 2020-08-19:
1) Embedded USB bootloader, so no need to buy a programmer to start with the project
2) USB Type C instead of USB Micro
3) Much faster, has FPU and more flash available so it becomes easier to implement features like force feedback

Have-Purple, 2021-09-08:
Would also like to show support for F4 support.  Quality Blue Pills are getting harder to find.

alekslnn, 2021-12-13:
Так что были работы с портом на st32f401???