FreeJoy-Team/FreeJoy #41: Improve latency

misteraddons, 2020-08-19:
I work on the MiSTerFPGA project. We have a testing suite used to measure input latency using an FPGA in the loop.

FreeJoy is something I'm very intersted in using in some upcoming projects, but it would be awesome to find a way to decrease the input latency. 

The best controllers have 0-5ms of latency, where FreeJoy currently has ~60ms:
`Average: 60.1666 ms
Max:     63.804 ms
Min:     56.408 ms
Std Dev: 1.591 ms
`
I set up all buttons to be BTN_GND, and ran the test with more than 2000 samples. 

Let me know if I can be of any assistance!

vostrenkov, 2020-08-19:
Hello,
i guess you get 60ms latency because of debouncing algorithm for the buttons. You can decrease debouncing time to improve input latency but there are ghost presses are possible at to low values

vostrenkov, 2020-08-19:
[image]

misteraddons, 2020-08-19:
Excellent, thank you! I haven't tested without debounce so it may still need to be tweaked but here are my current results:
[image]
Also, full spreadsheet of results here: [https://docs.google.com/spreadsheets/d/1KlRObr3Be4zLch7Zyqg6qCJzGuhyGmXaOIUrpfncXIM/edit?usp=sharing](url)

Looking forward to the possibility of 2P support :)

Avalonnw, 2020-08-19:
Ideally debounce should happen in the hardware. So, assuming it is, what would be the latency with software debounce off?

misteraddons, 2020-08-20:
The Notes column of the results shows that the top 4 results were with software debounce disabled.

Avalonnw, 2020-08-20:
> 
> 
> The Notes column of the results shows that the top 4 results were with software debounce disabled.

I can see that. I meant I was curious to see FreeJoy's latency with debounce off. I am aasuming it would be in 6.2-13.9 range?

misteraddons, 2020-08-20:
Sorry, I don't follow. Those are the results with debounce off.

Avalonnw, 2020-08-20:
FreeJoy results are with debounce set to 50ms. I says so in your table.

misteraddons, 2020-08-20:
That's only the bottom result. The other 4 are with "0 ms debounce".

Avalonnw, 2020-08-20:
> 
> 
> That's only the bottom result. The other 4 are with "0 ms debounce".

Ah, sorry, I thought the top 4 were done with other controllers. My bad.

misteraddons, 2020-08-20:
No problem :)

vostrenkov, 2020-08-20:
You can also try to set higher USB exchange rate in advanced settings. It should reduce input latency even more

misteraddons, 2020-08-27:
I ended up working with bootsector and he was able to get his own firmware down to 0.73 ms on MiSTer. If you ever want to improve latency, that might be something to look into. Cheers and thanks for the awesome, flexible code!