FreeJoy-Team/FreeJoy #42: Shift register vs. button matrix

franukis, 2020-08-25:
I am preparing a PCB that will be using Freejoy... I have a big number of buttons to be scanned, plus a couple rotary encoders and the full 8 analog axis, so I am concerned about the performance...

So, what would be a better idea for the buttons: 4 SPI 8-bit shift registers? Or as 6x5 button matrix with diodes?

Best regards

vostrenkov, 2020-08-25:
Hello,
The difference in performance shouldn't be noticеable. Matrix may be a bit faster but you will get 11 wires instead of 5 for shift registers.
30 is not a big number of buttons for freejoy, i'm trying to keep it fully functional with all 128 buttons and 8 axes enabled

franukis, 2020-08-25:
Great to hear that! :)

closed 2020-08-25 by franukis