FreeJoy-Team/FreeJoy #43: Dynamic HID Config

0xfgarcia, 2020-08-28:
Hi,

I am running into some trouble configuring the buttons.

HID Dynamic Enabled: Some buttons do not work.
All the buttons light up green in Freejoy physical buttons section, but only some show keystrokes in the Freejoy button log.

HID Dynamic Disabled: All buttons work.
All buttons work, light green and appear in the button log, but only 32 appear in Windows device settings.

[image]
[image]
[image]

vostrenkov, 2020-08-28:
Hi,
First of all do not use windows device settings for monitoring devices with more than 32 buttons as this tool support maximum 32 buttons. Use [VKB Button Tester](https://github.com/FreeJoy-Team/FreeJoyWiki/tree/master/3rd-party/software) or other direct input based tools.
Second thing: when you enable dynamic HID config it can break buttons order in the system, because it delete unused buttons and shift used buttons number.

So suggested testing plan for you:
1) Disable dynamic HID config
2) Run tester tool and check if all of your buttons work there
3) If everything is ok enable dynamic config 
4) Unplug and plug device again and restart tester tool
5) Check number of available buttons
6) If number matches to your config (92 buttons as i can see) check if buttons work

0xfgarcia, 2020-08-29:
Hi,

I don't see any difference in the number of buttons when I activate the dynamic HID. I have observed that in addition 12-position rotary switch also stops working. Dynamic HID disabled all 12 positions work in Freejoy & VKB, activated only 4 positions work and the Freejoy Configurator closes unexpectedly when I pass the fourth position of the switch.

Actually I don't have 96 pushbuttons, I have the following:

24x Matrix 4 x 6 Pushbuttons
4x Rotary Encoder Matrix (A and B pins Button Row, Center pin Button Column) E1 config in Freejoy wiki 
2x Axis Analog (Only 1 is configured for the 12P Rotary Switch Axes to Buttons)
1x 4 directional Switch + Encoder + Center Push Button

I get a total of: 51 buttons

Dynamic HID enabled:
[image]

Dynamic HID Disabled:
[image]

vostrenkov, 2020-08-29:
Ok, please save your config to file and upload somewhere. I will check it

0xfgarcia, 2020-08-29:
Thanks!

[STM32 v4.zip](https://github.com/FreeJoy-Team/FreeJoy/files/5145105/STM32.v4.zip)

vostrenkov, 2020-08-30:
Hi again!
First results of checking:
1) I see enabled "Minus" function for X axis that gives you output = X value - X value = 0. So axis doens work -> axis to buttons doesn't work either. Disable "Minus" function to get 12-position switch working
2) I got all buttons working in non dynamic mode (checked on v1.6.2 version as it is the latest version)
3) I confirm a bug with buttons mapping in dynamic mode, thank you for the report. It will be fixed in v1.6.3, till then please use non dynamic mode

vostrenkov, 2020-09-06:
Hi,
please check if v1.6.2b2 solve you issue

0xfgarcia, 2020-09-08:
Hi,

I have done a test with few buttons (12), with HID enabled, windows recognizes the number of buttons that I have configured in FreeJoy.

[image]

VKB Button Tester always indicates 128 buttons. Is the behavior correct ?

[image]

P.S: The new green bar in axes to buttons is very cool, great idea.
[image]

vostrenkov, 2020-09-08:
VKB tester always shows 128 buttons, i have forgotten about that. So it is ok if you get correct number in windows game controller panel

0xfgarcia, 2020-09-08:
Ok, thanks!

Solved.

closed 2020-09-08 by 0xfgarcia