FreeJoy-Team/FreeJoy #44: Encoder Type

0xfgarcia, 2020-08-28:
Encoders work 6x but I can't select encoder type.

[image]

vostrenkov, 2020-08-28:
Please specify your encoder part number

0xfgarcia, 2020-08-29:
My fault. I was with Freejoy Configurator v1.6.0b5, I can select the encoder type with v1.6.2.b0 but I can't get it to work correctly.

The strange thing is that with 1.6.0b5 the encoder rotation were recognized by the FreeJoy Configurator & VKB but with the current version I can't get them to work, I have multiple unwanted pulses.

I don't know if it will be related but when connecting an stm32 configured with the Freejoy Configurator v1.6.0b5 in the Configurator v1.6.2b0 encoders function appear as POV Left & Down instead of Encoder A and Encoder B in Button Config. I have changed them to Encoder A and Encoder B but they don't work for me.

4x Rotary Encoder Matrix (A and B pins Button Row, Center pin Button Column) E1 config in Freejoy wiki

[image]

vostrenkov, 2020-08-29:
Configurator version must match firmware version (except of number after "b" symbol). Otherwise board does not accept written config

closed 2020-08-30 by 0xfgarcia