FreeJoy-Team/FreeJoy #45: How do I build from source

GregWoods, 2020-08-30:
I would like to experiment with the code myself, but I am not sure where to start. I have experience of Arduino, PlatoformIO and some Rust experience on STM32F1 and F4, but I need a little direction on where to start with this project.
What IDE and compiler do you use, and how is it set up?
I notice a makefile, MDK-ARM folder, but also armgcc folder.

Do I need the ARM MDK software? If so which version? Can I configure VS Code and gcc?

Any help would be greatly appreciated.

vostrenkov, 2020-08-30:
Hi, 
MDK ARM suppose Keil uVision v5. Keil project is placed in MDK ARM folder.
It should be possible to build with gcc and makefile placed in armgcc folder but it is broken now. Prerelease v1.6.3 we be ready very soon, gcc makefile will be fixed there

GregWoods, 2020-08-30:
Now that was a fast response! Thank you
Would the free version if uVision work? It has a 32Kb limit. I assume the limit is compiled code size.

vostrenkov, 2020-08-30:
With -O3 or -Os optimization it should fit 32Kb. 
Also notice there 2 targets: application and bootloader. Bootloader must be flashed first, otherwise application wont work

vostrenkov, 2020-10-17:
In v1.6.2b4 makefile for building with GCC is fixed. Manual for building is placed in **armcc** folder of the project

delpitec, 2023-08-26:
What are the steps to run makefile (windows) ?

mupfdev, 2023-08-31:
Would there be interest in migrating to CMake? I could actively support the project here.

This would allow the project to be compiled and debugged independently of the platform and IDE.