FreeJoy-Team/FreeJoy #46: Analogue Axis to Buttons - Leo Bodnar 12 position switch

snubbsyuk, 2020-09-03:
Hi,
I'm getting odd behaviour when using the Axis to Buttons feature with a Leo Bodnar 3-pin 12 position switch.
Basically, I'm assigning switch positions 1-5 as "Shift" buttons so that 2 rotary encoders can serve multiple functions.
It works 70% of the time, however, when turning the rotary (e.g. logitech button 14/15) it'll report button 14 pressed, unpressed, pressed unpressed, then out of no-where it'll act as if the rotary switch has been changed for a split second & engage a different shift-version of the encoder momentarily.

Reviewing the axis RAW output, it appears to stutter by about 15 points, every second or so, and about 100 points when other buttons/rotaries are engaged.

Would it be possible to provide a wider band for each "position" when using the Axis to Buttons feature? I'm only speculating that this is the issue, so I could be wrong but would appreciate your support.

Thanks

vostrenkov, 2020-09-03:
Hi, 
Please save your config to file and upload somewhere I could get it

snubbsyuk, 2020-09-03:
Thanks for taking a look vostrenkov
I've uploaded my config to https://github.com/snubbsyuk/DimSimConfig/blob/master/DimSimR1-12pos.conf
It's a reduced config with only a few buttons assigned. The single encoder I'm testing with here is on physical button 19/20, the rotary switch is on physical buttons 1-12 (using only 1-5 for Shift function)

Cheers

snubbsyuk, 2020-09-03:
An example of the output I'm getting (in shift position 1) when moving the encoder back and forth
21:44:43: Logical button №11 pressed
21:44:43: Logical button №11 unpressed
21:44:43: Logical button №11 pressed
21:44:44: Logical button №11 unpressed
21:44:44: Logical button №11 pressed
21:44:44: Logical button №11 unpressed
21:44:44: Logical button №10 pressed
21:44:44: Logical button №12 pressed
21:44:44: Logical button №14 pressed
21:44:44: Logical button №12 unpressed
21:44:44: Logical button №14 unpressed
21:44:44: Logical button №10 unpressed
21:44:44: Logical button №11 pressed
21:44:45: Logical button №13 pressed
21:44:45: Logical button №15 pressed
21:44:45: Logical button №11 unpressed
21:44:45: Logical button №13 unpressed
21:44:45: Logical button №15 unpressed
21:44:45: Logical button №11 pressed
21:44:45: Logical button №11 unpressed
21:44:45: Logical button №10 pressed
21:44:45: Logical button №10 unpressed
21:44:45: Logical button №10 pressed
21:44:45: Logical button №10 unpressed
21:44:45: Logical button №10 pressed
21:44:45: Logical button №10 unpressed
21:44:46: Logical button №11 pressed
21:44:46: Logical button №13 pressed
21:44:46: Logical button №15 pressed
21:44:46: Logical button №11 unpressed
21:44:46: Logical button №13 unpressed
21:44:46: Logical button №15 unpressed
21:44:46: Logical button №11 pressed
21:44:46: Logical button №11 unpressed
21:44:46: Logical button №11 pressed
21:44:46: Logical button №11 unpressed
21:44:46: Logical button №10 pressed
21:44:46: Logical button №10 unpressed

in shift position 2:
21:45:27: Logical button №12 pressed
21:45:28: Logical button №12 unpressed
21:45:28: Logical button №12 pressed
21:45:28: Logical button №12 unpressed
21:45:28: Logical button №13 pressed
21:45:28: Logical button №13 unpressed
21:45:28: Logical button №12 pressed
21:45:29: Logical button №10 pressed
21:45:29: Logical button №14 pressed
21:45:29: Logical button №10 unpressed
21:45:29: Logical button №14 unpressed
21:45:29: Logical button №12 unpressed
21:45:29: Logical button №11 pressed
21:45:29: Logical button №13 pressed
21:45:29: Logical button №15 pressed
21:45:29: Logical button №11 unpressed
21:45:29: Logical button №15 unpressed
21:45:29: Logical button №13 unpressed
21:45:29: Logical button №12 pressed
21:45:29: Logical button №12 unpressed
21:45:29: Logical button №13 pressed
21:45:29: Logical button №13 unpressed

vostrenkov, 2020-09-04:
Ok, checked.
I will track this as bug. Should be fixed in 1.6.3, so stay updated

snubbsyuk, 2020-09-04:
> Ok, checked.
> I will track this as bug. Should be fixed in 1.6.3, so stay updated

Thank you!

vostrenkov, 2020-10-17:
Hi, please try version 1.6.2b4

snubbsyuk, 2020-10-17:
I've tested 1.6.2b4 and finished configuring my setup (I think I actually refreshed the page just as you uploaded it :D ) 
All is working well now with this version. Thanks for this functionality!

closed 2020-10-17 by snubbsyuk