FreeJoy-Team/FreeJoy #47: Black pill?

rewsenj, 2020-09-06:
Hi I wanted to know if you plan on making this compatible with the STM32 black pill?

vostrenkov, 2020-09-06:
Hi, it is compatible with any board with STM32F103. 
I know there are some bluepills boards with STM32F4 chips, they are planned to be supported later for FFB version of the software

bayerla, 2020-09-06:
Working [https://github.com/FreeJoy-Team/FreeJoyConfigurator/issues/17](url)

bayerla, 2020-09-06:
working [https://github.com/FreeJoy-Team/FreeJoyConfigurator/issues/17#issue-655216396](url)

bayerla, 2020-09-06:
[image]

rewsenj, 2020-09-07:
Ah yep, turns out I was being exceedingly dumb and neglected to click 'write config to device'... Thanks for the quick reply.

I have some other questions:

1: I've connected two different button matrices to the board but the software is reading them as one single matrix, is there any way to make it see them as separate matrices?

2: I've been struggling to connect a single button(as in one solitary button as opposed to a matrix)  to the board using either vcc/btn gnd or btn gnd/btn vcc. whenever I press the button the indicator doesn't change to green. It's not a fault with the button, I checked.

3: I've been trying to connect a hall sensor but it doesn't seem to be working, when I wave a magnet around it nothing happens. The sensors are [A1324LUA-T](https://www.digikey.com/product-detail/en/allegro-microsystems/A1324LUA-T/620-1432-ND/2728144). I tried with and without a 47µF capacitor (maybe too large capacitance?) but got nothing. I tried with a pot and it worked fine. Looking back at the website it says it needs a 4.5~5.5V supply so I probably just answered my own question. do you think [this](http://stm32duinoforum.com/forum/viewtopic_f_3_t_4176.html) is a viable solution?

Sorry I'm pretty new to this.

vostrenkov, 2020-09-07:
1) No, only one matrix supported
2) Show your pins config or upload your config file. Do you have dynamic HID enabled?
3) Yes voltage is a possible reason. Try with 5 volts but you should use a voltage divider on signal to not exceed 3.3V on analog pin

rewsenj, 2020-09-07:
Dynamic HID wasn't enabled, so I enabled it, but it still doesn't work.

[image]

vostrenkov, 2020-09-07:
I have it working with this config on v1.6.2b2.
Make sure you connect buttons right - in case of Button_VCC it should short pin to 3.3V or in case of Button_GND to GND

rewsenj, 2020-09-07:
ah yep, the button is working now thx

It says [here](https://forum.hobbycomponents.com/viewtopic.php?t=2422) that the serial pins are 5v tolerant so is it fine to connect them without a divider (a3 and a4)?

vostrenkov, 2020-09-07:
The picture also shows that a0-a7 pins are not 5V tolerant.
Also if you will get  more than 3.3V on axis input it will be counted as maximum value. So you can connect sensor supply to 5V but you have to make sure its output will fit 0 - 3.3V range

rewsenj, 2020-09-07:
I'll get some Bidirectional LLCs

vostrenkov, 2020-09-07:
Logic levels converters won't fit if you need analog input because they have logic output. Easiest way is to use voltage divider with 2 resistors

rewsenj, 2020-09-07:
I'll get resistors then. 
I only have 10uF capacitors, is that too high or might they work?

vostrenkov, 2020-09-07:
10uF between supply and gnd should be fine

rewsenj, 2020-09-09:
haha it works! right now i'm using 1k and 2k resistors but I can go higher (up to 100k and 200k). what would you recommend?

vostrenkov, 2020-09-09:
I'd recommend to stay in 10K - 100K range

rewsenj, 2020-09-20:
Ok, I've managed to get it working. Mostly. The axis are connecting to the computer just fine but it doesn't seem to register the button presses (while freejoy configurator does). Any idea why?

vostrenkov, 2020-09-20:
Have you done logic buttons setting according to wiki?

rewsenj, 2020-09-21:
I've rewired it so there is only one matrix but now it's doing this:
[image]
[image]

When I look at it in game controller it doesnt read any button presses exept when I press button 1 (lefthat1) where it reads it as all buttons being pressed

vostrenkov, 2020-09-21:
This is exactly what it should do with configuration on your pic.
On buttons tab you should map your physical buttons (left part of the screen) to your logical buttons (right part of the screen). Logical buttons is what your game read, physical buttons is what you have connected to the controller, so phy.btns work as source for log.btns
Now you set all your logical buttons mapped to physical button 1, so all them triggered when you press phy.btn1

rewsenj, 2020-09-21:
Oh, it all makes sense now. Everything is working. Thank you so much for your help, I wouldn't have gotten this far without it.

closed 2020-09-21 by rewsenj

pierre-vo, 2021-01-02:
@rewsenj does your black pill have a STM32F4 or STM32F103?

makermindlane, 2021-09-29:
@vostrenkov I'd love to help bring support for stm32f4. Would be tremendous help if i could get a headstart or information about project architecture. :)

vostrenkov, 2021-11-16:
> @vostrenkov I'd love to help bring support for stm32f4. Would be tremendous help if i could get a headstart or information about project architecture. :)

Work on porting to F4 is in progress. I believe it will be ready quite soon

Monkeyja, 2024-06-28:
I want to know you have plans for implement ffb function or not 😅