FreeJoy-Team/FreeJoy #49: FreeJoy is workin but DCS dosen't recognize a button press.

Sparkowl, 2020-09-08:
The FreeJoy Board is recognized by Win10, all Buttons (via shift registers) and axes do work in the configuration app and in the system settings. When try to bind a button in DCS or in IL2 it dosen't recognize a button press. After starting up DCS with open System setting dialog for gamecontroller it showes no pressed button in there either.
My board does have the 10KOhm as R10 Resistor soldered. 
Thanks for your outstanding work.

vostrenkov, 2020-09-08:
Hi,
If your controller operates in system control panel the problem is not with the device.
I'm not really familiar with DCS but I have two devices on freejoy working in IL2.
Once I had problems with mapping inputs in this game but it was solved by clearing /input/devices.txt file in game folder

Sparkowl, 2020-09-08:
Thank you for your quick reply. I will try clearing the devices.txt

Sparkowl, 2020-09-08:
I had the Freejoy connected over an USB 3.0 Hub. The problem disappeared after connecting it directly to the PC. Are there any limitations in conjunction with Hubs?

vostrenkov, 2020-09-08:
I'm not sure, a lot may depend on used hub. I'm glad you managed to find a reason

Sparkowl, 2020-09-08:
Thank's for your support and keep up the good work.

closed 2020-09-08 by Sparkowl