FreeJoy-Team/FreeJoy #50: help

majidcactus2020, 2020-09-09:
Dear Sir  
Can STM32F103ZET6 be used in this program?

vostrenkov, 2020-09-10:
Hi,
I don't see a reason why not but I have not tried that.
I can try with my f103nucleo board but i need some time to get to it

IIIIIoppIIIII, 2020-09-10:
Hi! 
Is it possible to enter more than six characters in the USB custom name field?

vostrenkov, 2020-09-10:
> Hi!
> Is it possible to enter more than six characters in the USB custom name field?

Hi, yes it supports up to 20 characters, but it is a bit bugged now. There are some troubles in entering device name after reading config back

vostrenkov, 2020-10-17:
fixed in configurator

closed 2020-10-17 by vostrenkov

majidcactus2020, 2020-10-19:
Dear sir,
 I know little about programming & software.
  STM32F103ZET6 has 144 -pin packages .
A higher number of keys and buttons can be used. and  keys  ,buttons, I use
for all aircraft.

I hope to near you very soon

Your faith fully

 Majid Eslamian

vostrenkov, 2020-10-19:
There is limit of 128 buttons and 8 axis per usb hid device in windows. You can easily achieve 128 buttons on F103C8T6 using matrix connection or shift registers periphery