FreeJoy-Team/FreeJoy #51: Feature request: Fast encoder as  mouse

robcazzaro, 2020-09-13:
I just discovered FreeJoy, really amazing.

I was planning to build my own USB interface for Retropie/Mame. A single Blue Pill with USB HID implementing both a joystick and optical rotary encoder. The rotary encoder to be used for old arcade games like Tempest and Arkanoid. 

Usually all optical encoders for arcade emulators work as a mouse, sending an x-axis (or y-axis) increment for every encoder step. I see that FreeJoy sends button events instead, because it's been implemented as a joystick profile.

Do you think you will ever have any interest in adding a USB mouse HID profile and using encoders as a mouse? It would also work as trackball (but only if there is a way to implement another fast encoder for the other axis)

vostrenkov, 2020-09-14:
Hi, 
I'm going to add support for mouse but not sure how it will be implemented yet.
I will back to this question when I start development of this feature

robcazzaro, 2020-09-14:
Great, I'll keep an eye on this. 

Other solutions simply send a Mouse.move(1, 0) when the encoder increments by one, Mouse.move(-1, 0) when the decoder decrements by one.

I tried putting together a mouse library and an encoder library, and the simple code above works just fine

vostrenkov, 2020-09-14:
I want to bind mouse to axis to be able to control by an analog ministick. Encoder also ok but I haven't thought yet how all these features should be implemented