FreeJoy-Team/FreeJoy #52: Toggle switch connections

franukis, 2020-09-14:
Hi! I am finishing a PCB to support a bluepill with Freejoy and I am a little lost with the configurator options for toggle switches:

I understand I have to configure them as buttons and then select toggle but

- For a 3 pos toggle (think Flaps: Full-Half-Auto) do I need an ON-ON-ON? Or can I do with a ON-OFF-ON? What would be the diff in configuration? I was thinking to use on-off-on mapping only Full and Auto...would that work?

I've browsed the wiki but could not find this info. I think some examples of connection+their configs would be helpful (I can gladly help creating them once I get it clear myself)

vostrenkov, 2020-09-14:
Hi,
Do you need 3 certain positions, right?
In case of ON-OFF-ON you can set two physical buttons that correspond pressing positions 1 and 3.
Then set 4 logical buttons:
1) button 1 toggle on for function 1 (Auto)
2) button 2 togge on for function 3 (Full)
3) button 1 toggle off for function 2 (Half)
4) buttob 2 toggle off also function 2 (Half)
You will need to map both buttons 3 and 4 for "Half" function in your game

Another option is to use ON-ON-ON switch directly to have 3 physical buttons for 3 logical states

franukis, 2020-09-14:
Very clear, thanks a lot!

franukis, 2020-10-05:
Hi again,

after reconsidering your reply, I have some doubts:

- As most ON-ON-ON switches seem to be double pole and, hence bigger in size, I wuld rather stick to the single pole ON-OFF-ON... However, mapping 2 physical connections to 4 logical buttons seems problematic (could be that some PC interface does not allow to map the same function to two different controller buttons). Would it be possible to map the 2 physical connections to 3 logical buttons? Like this:

- Phys. conn 1 ON --> Logic button 1
- Phys conn 1 OFF --> logic button 2
- Phys. conn 2 ON --> logic button 3
- Phys. conn 2 OFF --> logic button 2

Of course, the program would have to poll the connection, and remeber the previous state, to recognize both possible transitions and map them to the same logic button pulse (both ON1-->OFF and ON2-->OFF), but it should be doable... What do you think?

vostrenkov, 2020-10-05:
This is not possible in this software unfortunately.
Software mapper software may help you with mapping to buttons from USB device to one virtual button

franukis, 2020-10-07:
Is it a hard limitation of the SW? I mean, I could think of setting a parameter on the configurator to tell there are 2 logic buttons "coupled". Then, in the FW, if there are any coupled buttons, we add a layer to remap before the usb report, reducing the number of buttons, and mapping the 2 coupled buttons to a single one of the usb report...

Of course, I understand this is not done at the moment, but seems not too hard to implement, as an option, and could be interesting. What do you think?

BR

vostrenkov, 2020-10-07:
Not hard logically but quite hard to keep in accordance with current architecture.
You will need to add extra button to buttons configuration and handling structures or unpin logical button number from buttons array index. This require to rewrite tons of code
Maybe a solution is to create some kind of scripting system but I think it is "overkill" for now and you can use external scripting/mapping software if needed

vostrenkov, 2020-10-07:
I have an idea how to make 3 pos toggle switch from ON-OFF-ON switch with no software modification but with easy hardware mod.
Connect edge pins of the switch to GND and VCC and make a voltage divider between them with center point connected to central (OFF position) pin.
Then you can you "axis to button" connected to central pin for 3 physical buttons

vostrenkov, 2020-10-07:
[image]

franukis, 2020-10-07:
It is indeed a good idea. Thanks for your comments, didn't want to sound pushy, just thinking about possibilities 😅

Thanks for all the hard work!!

FSA2001, 2023-09-11:
Hello, I am in the first steps of putting together an accelerator command for X Plane 11 with STM32
Is there a code for that or does it have to be created?
Thank you

lttlrck, 2025-01-01:
> [image]

I've just been puzzling over this and found this elegant solution! Thanks!