FreeJoy-Team/FreeJoy #53: logitech G25/27 shifter

yogggoy, 2020-09-17:
Hello. Thanks for great project.
Q: It is possible to config FreeJoy for support logitech shifter? some manual how do this?
I would like to use a shifter and handbrake on the same bluepill if possible.
thanks!

vostrenkov, 2020-09-17:
Hi, 
I believe it is possible since shifter and handbreak is just a bunch of buttons and axis. However I can not give you an exact guide because i never did that

yogggoy, 2020-09-18:
i am try connect g27 gear shifter to freeJoy Bpill.

All buttons and axis were found well. Wiki have great description for config Axis to buttons. But i don't understand how to config 2 axis for buttons matrix? 

pinouts and schem found at:
https://github.com/robotsrulz/SP_Adapter
https://github.com/chrisjansson/Shifter

G27 gear shifter it is 2 potenciometer, used as XY plot.
And reverse speed in same position as 6speed, with pushed button.
```
X axis
^
| 1    3     5
|
|    neutral
|
| 2    4     6/R
|-------------------> Y axis
```

it is possible to config, or need go to modify src?

vostrenkov, 2020-09-18:
I guess you will need 3x3 matrix including neutral.
Shifts functionality may help you with your configuration:
1) Create two pots axis with 3 buttons from each
2) For axis Y define 1st and 3rd buttons as shifts on buttons tab
3) Configure buttons for your rows and set corresponding shifts

vostrenkov, 2020-09-18:
```
          X axis
          ^
(Shift 2)  | 1    3     5
           |
           |    neutral
           |
(Shift 1)  | 2    4     6/R
            -------------------> Y axis
             B1   B2   B3
```

Button set as Shift 1 activates lower row, Shift 2 activates upper row, middle position of X axis is neutral

yogggoy, 2020-09-18:
Great! I go try it.
And one more question, can i use two Shift_btn on one logical_btn?
This logitech shifter have Reverse speed at same position a 6 speed, but with pushed gear knob (like a Z axis)
```
          X axis
           ^
(Shift 2)  | 1    3     5
           |
           |    neutral
           |
(Shift 1)  | 2    4     6/R
            -------------------> Y axis
             B1   B2   B3
                       B8
```
Reverse - B3, [Shift1 and B8]

vostrenkov, 2020-09-18:
Reverse mat cause some issues, right. 
In FreeJoy shifts are logical buttons so you can set shift on another shift. I think that should work for you, at least you can try do to the trick

yogggoy, 2020-09-27:
I have some trouble with "FreeJoy Shift1/2" scheme on "G27 gear-shifter".
I try debug and build project on keil 5.31, but current project is so big for free version Keil.
Optimization O2 (other O1 O3 is bigger)
```
FreeJoy\FreeJoy.axf: error: L6047U: The size of this image (33816 bytes) exceeds the maximum allowed for this version of the linker
Finished: 0 information, 0 warning, 0 error and 1 fatal error messages.
"FreeJoy\FreeJoy.axf" - 1 Error(s), 0 Warning(s).
```

redefine MAX_..._NUM in file `common_defines.h` broke build, 
and fast monkey patching dont help me (just delete `out of rangne` errors)
FreeJoy Configurator cant understand my bin 

it is possible to disable some functions (e.g. encoders) in code ?

the problem is - some `gear` is lost after 1-5 minuts.

vostrenkov, 2020-09-27:
You can build it with gcc with no code size limitations. Makefile for both app and bootloader placed in gcc folder.

I dont recommend to edit axes/button/other stuff cnt definitions. If you need to reduce the code size -Os option should be fine (or -O3 in MDKARM v5)
Share your configuration, I will check whats wrong

yogggoy, 2020-09-27:
Try build on ubuntu-WSL. `build-essential` is installed:
```
nyah@DESKTOP-C5IKDBM:~/projects/FreeJoy/armgcc$ make
make: *** No rule to make target 'build/analog.o', needed by 'build/FreeJoy.elf'.  Stop.
```

config file for only [6 speed] logical buttons:
https://gist.github.com/yogggoy/93346f7d73138f07237d6757825a10b0

[image]

I'll try to record a video with a problem/

yogggoy, 2020-09-27:
https://youtu.be/fqw6FA2M4bc
also video has description

yogggoy, 2020-09-27:
Found only one small problem without debugging.

https://github.com/FreeJoy-Team/FreeJoy/blob/1029b5d54c10b40f85b9484309e5e35c5b335df2/application/Src/buttons.c#L896

`shifts_state` used before assigned. line 769, 808, ...

L897: may be rewriten:
```
	for (uint8_t i=0; i<5; i++)
	{
		if (p_dev_config->shift_config[i].button >= 0)
		{
			shifts_state |= (logical_buttons_state[p_dev_config->shift_config[i].button].current_state << i);
		}
	}
```
But that is not solve trouble, just little optimization.

vostrenkov, 2020-09-27:
Will take a look a bit later.
Yeah, buttons code is extremely not optimized

vostrenkov, 2020-09-30:
I noticed you set offsets for your axes. They may cause undesired behaviour for analog sensors and mostly used for digital hall sensors. So I'd recommend you to use zero offset values.

I also made a video of setting up 6 gear shifter: https://youtu.be/3nYNJg64rMU
I i have it working stable with this setup

KarlSteeg, 2025-05-04:
Howdy.
Has anyone gotten the reverse gear working?