FreeJoy-Team/FreeJoy #54: Update firmware from linux

Ntemis, 2020-09-18:
Hi, congratulations for this excellent firmware project.
Is there any chance we can get the configuration and updater app work in a linux enviroment like debian/ubuntu?

vostrenkov, 2020-09-18:
QT based configurator coming soon. I believe it will work on Linux

Reksotiv, 2020-09-18:
You can try https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/releases , readme inside.

Ntemis, 2020-09-18:
execv error: No such file or directory

Reksotiv, 2020-09-18:
what linux do you use? clean ubuntu 20.04 https://i.imgur.com/9BjBbqo.png

Ntemis, 2020-09-18:
20.04 Mate

Reksotiv, 2020-09-19:
Clean mate https://i.imgur.com/pzpBZPz.png . Can you extract https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/releases/download/0.4.6/TEST-Linux-FreeJoyQt-v1.6.2.tar.xz  and run ./AppRun in terminal inside AppDir?

Reksotiv, 2020-09-19:
I repeated the error in Linux Mint. Can you test new version https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/releases ?

Ntemis, 2020-10-25:
It works but it cant see the device. I did push the 99-hid-FreeJoy.rules per instructions

Ntemis, 2020-10-25:
btw i am seeing these in dmesg
[ 2511.220727] usb 3-1: new full-speed USB device number 17 using uhci_hcd
[ 2511.439751] usb 3-1: New USB device found, idVendor=0483, idProduct=5750, bcdDevice= 2.00
[ 2511.439754] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2511.439755] usb 3-1: Product: FreeJoy
[ 2511.439757] usb 3-1: Manufacturer: STMicroelectronics
[ 2511.455509] hid_map_usage: 38 callbacks suppressed
[ 2511.455510] STMicroelectronics FreeJoy: Invalid code 768 type 1
[ 2511.455512] STMicroelectronics FreeJoy: Invalid code 769 type 1
[ 2511.455513] STMicroelectronics FreeJoy: Invalid code 770 type 1
[ 2511.455514] STMicroelectronics FreeJoy: Invalid code 771 type 1
[ 2511.455515] STMicroelectronics FreeJoy: Invalid code 772 type 1
[ 2511.455516] STMicroelectronics FreeJoy: Invalid code 773 type 1
[ 2511.455517] STMicroelectronics FreeJoy: Invalid code 774 type 1
[ 2511.455517] STMicroelectronics FreeJoy: Invalid code 775 type 1
[ 2511.455518] STMicroelectronics FreeJoy: Invalid code 776 type 1
[ 2511.455519] STMicroelectronics FreeJoy: Invalid code 777 type 1
[ 2511.455582] input: STMicroelectronics FreeJoy as /devices/pci0000:00/0000:00:1a.0/usb3/3-1/3-1:1.0/0003:0483:5750.0007/input/input27
[ 2511.512855] hid-generic 0003:0483:5750.0007: input,hidraw3: USB HID v1.10 Joystick [STMicroelectronics FreeJoy] on usb-0000:00:1a.0-1/input0

Reksotiv, 2020-10-26:
[ 2511.439751] usb 3-1: New USB device found, idVendor=0483, idProduct=5750, bcdDevice= 2.00
5750 - old PID, does not work in some linux distributions. In new version we forgot to change it to 5757 in firmware. 
You need to change PID in windows or wait for new firmware
[image]

Or you can try edit 99-hid-FreeJoy.rules. Change 
KERNEL=="hidraw*", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5757", MODE="0666"
to and reboot PC
KERNEL=="hidraw*", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5750", MODE="0666"
but this does not guarantee work (works in linux mint does not work in ubuntu)

closed 2020-12-03 by Ntemis