FreeJoy-Team/FreeJoy #55: Support for 74HC595 shift register.

rafmonteiro, 2020-09-22:
It seems that the 74HC165 is now discontinued and it's becoming hard to find it (I ordered a few and it will take 2 months to arrive in Canada). Is there any plans to support  74HC595?

vostrenkov, 2020-09-22:
Hi,
165 are input registers and can be used for connecting buttons, but 595 are output registers and can be used for connecting leds for example.
So if you can not find 165 in stock try with CD4021which also supported in freejoy

closed 2020-10-03 by rafmonteiro

bayerla, 2020-10-03:
No Problems to get it in Germany, If at November you still have problems I could try to sent you some.....
[https://www.reichelt.de/shift-register-3-state-ausgang-2--6-v-dil-16-74hc-595-p3269.html](url)

FlisherOfatale, 2021-03-22:
I don't think the 74HC165 is going anywhere, there are plenty of them available on digikey.ca, SMD, SMT, etc...