FreeJoy-Team/FreeJoy #57: (｡･∀･)ﾉﾞHi Does FreeJoy support a 1/4 coder like CTS 288?

Bliss233, 2020-09-27:
(｡･∀･)ﾉﾞHi
Does FreeJoy support a 1/4 coder like CTS 288?

vostrenkov, 2020-09-27:
Hi,
Yes, 1/4 encoders support present

Bliss233, 2020-09-27:
> 嗨，
> 是的，目前支持1/4编码器

\(^o^)/~ Great, do I need to set up the encoder to use CTS288?I want to install 14 buttons and 5 encoders. Can the encoders be connected in parallel?

vostrenkov, 2020-09-27:
We have wiki manual how to set up the controller:
https://www.github.com/FreeJoy-Team/FreeJoyWiki/tree/master/eng%2FEncoders-connection.md

Bliss233, 2020-09-27:
[image]
[image]

Thank you very much. Can I connect the line this way?

vostrenkov, 2020-09-27:
I think yes, but you will need to add diodes in this matrix
Better try it on devboard first

Bliss233, 2020-09-27:
Thank you for your suggestion. After the encoder is connected, set them as Button ROW and Button Colum, so you can use 1/4 of the encoder for 2 buttons.

Bliss233, 2020-10-07:
Hello, I'm having some problems, the encoder turns to 00 10, 12, 02, 00 after rotating to the left, and 00 02, 12, 10, 00 after rotating to the right. I use CTS 288 encoder 1/4, may I ask where to set it?
[image]
[image]
[image]

Bliss233, 2020-10-08:
I need to rotate the encoder 4 times to press the 1 button

snubbsyuk, 2020-10-11:
> I need to rotate the encoder 4 times to press the 1 button

Hey, you need to go into the "Encoder" tab and select "Encoder_4x"
I use CTS 288 encoders and they work correctly with the above setting.

Bliss233, 2020-10-11:
> > 我需要旋转编码器4次以按1按钮
> 
> 嗨，您需要进入“编码器”选项卡并选择“ Encoder_4x”。
> 我使用的是CTS 288编码器，它们在上述设置下可以正常工作。

I set the option of 4X, but it still needs to rotate 4 times to light up the button. Can I have a look at your Settings?

vostrenkov, 2020-10-11:
> > > 我需要旋转编码器4次以按1按钮
> > 
> > 
> > 嗨，您需要进入“编码器”选项卡并选择“ Encoder_4x”。
> > 我使用的是CTS 288编码器，它们在上述设置下可以正常工作。
> 
> I set the option of 4X, but it still needs to rotate 4 times to light up the button. Can I have a look at your Settings?

Please show your settings on encoders tab. Is it possible you set resolution for wrong encoder or haven't written settings to the controller?

snubbsyuk, 2020-10-11:
> > > 我需要旋转编码器4次以按1按钮
> > 
> > 
> > 嗨，您需要进入“编码器”选项卡并选择“ Encoder_4x”。
> > 我使用的是CTS 288编码器，它们在上述设置下可以正常工作。
> 
> I set the option of 4X, but it still needs to rotate 4 times to light up the button. Can I have a look at your Settings?

I've found I have to write the config to the device, unplug & replug USB for this kind of functionality to take effect.

closed 2020-11-18 by vostrenkov