FreeJoy-Team/FreeJoy #58: Antivirus

kumrik, 2020-10-01:
Hi, my antivirus (Norton) deleted all .exe files automatically after unpacking. It says "setup.exe contained threat WS.Reputation.1". Is this project safe?

vostrenkov, 2020-10-02:
Hi, 
which version did you get? current 1.6.2 version does not have any setup files

kumrik, 2020-10-02:
Hi, I download two .rar files. Freejoy-master and FreeJoyConfigurator-master. All deleted files are in folder FreeJoyConfigurator-master/FreeJoyConfigurator/publish. Name deleted files: HidLibrara.dll; InWit.WiewModel.Ulits.dll; FreeJoy Configurator.exe; setup.exe

vostrenkov, 2020-10-02:
Please use software from releases as described in wiki: https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Required-software.md

kumrik, 2020-10-02:
Ok, thank you for info

closed 2020-10-17 by vostrenkov