FreeJoy-Team/FreeJoy #61: hx711

M0rta, 2020-10-16:
Hi!
Support load cells?

vostrenkov, 2020-10-16:
Hi,
You can connect load cells to analog input through the amplifier

M0rta, 2020-10-16:
ok! great, but need to use 2 analog inputs?
the arduino way looks like this:

#define LOADCELL_DOUT_PIN A0
#define LOADCELL_SCK_PIN A1

same way in freejoy?
tnx!

vostrenkov, 2020-10-16:
Please take a look to our wiki page. In FreeJoy you do everything through configurator utility

M0rta, 2020-10-16:
ok, believe me I'm looking for what I need but I can't find how to connect it, and that's why I'm asking this way.
my doubts are that in arduino, I connect + | - | and 2 pins, since there are 4 cables and only 3 in an analog.
I go back to check the wiki, but if you can please give me some help, I really appreciate it.

vostrenkov, 2020-10-16:
Since you have SCK and DOUT pins for Arduino you use not analog but digital interface. I guess it is hx711 IC that amplify analog signal from load cell and covert it into digital format.
FreeJoy doesn't support HX711's digital interface so if you want to use load cells with it you should use analog signal (on pins A0-A7). Raw analog signal from load cell is quite small and you will have to use amplifier (like AD623) between load cell and bluepill analog pin

vostrenkov, 2020-10-16:
All analog sensors are pretty the same for freejoy so amplifier output connects to analog pin just like pot or hall sensor

M0rta, 2020-10-16:
I understand perfectly now.
it was never going to work for compatibility reasons.
Without support for hx711, I will not be able to connect the load cell, if I want to use freejoy.
Thank you very much for your work and your time.

closed 2020-11-18 by vostrenkov