FreeJoy-Team/FreeJoy #62: Manual on tuning encoders

blukk, 2020-10-16:
I tried several encoders, debounced by hardware, connected via diodes but I still cannot get them to work properly. In MMJOY they do work fine but in Freejoy I seem to miss maybe the on/off function? Or is there another way to tune the settings. They now still output erratic movements, stop responding when turning a bit faster and skip and ghost a lot.
Maybe I am just not tuned them right in Freejoy. Is there some kind of manual or trick to get the settings right to make them perform optimum?
Respect for this software and Thanks;)

vostrenkov, 2020-10-16:
Ok, I got it. I will try to connect encoder to mmjoy and compare the output to figure out the issue

blukk, 2020-10-28:
One other question, I added a few potentiometers directly to the blue pill, and a bit of jitter and noise are shown as expected. However when going to the windows calibration wizard the axis will calibrate correctly but the blue calibration bar per Axis is flickering,  raw values jump in large values. It is impossible to see the axis moving on the blue bars contrary to for example what I have seen with an Arduino board. is there a way to filter the, I guess noise, as much in the software so the calibration toolbars under the windows calibration tool will shop up stable? I tried Deadband and filter as scaling but that doesn't work.
Thank you for your support ;)

blukk, 2020-12-08:
> Ok, I got it. I will try to connect encoder to mmjoy and compare the output to figure out the issue

Hello. Did you ever got the encoder function optimized. I got it to work somewhat properly but for instance, using it as  heading bug rotary in flighsim software is still impossible. Whenever I turn the encoder very slow it will move up to speed but turning a bit faster and the movement of the heading bug stagnates instead of speeding up. I play with the delay- and press timers a lot but none seems to make any difference. I also don't notice any changes when selecting different encoder types.  In MMjoy2 I have only 1 timer setting available but this is exactly the one that tunes it perfectly. How can I do this similar in Freejoy?

Thanks very much and keep up the good work, highly appreciated.