FreeJoy-Team/FreeJoy #63: Where is the firmware file for 1.6.2 b3?

uwlftoe17, 2020-10-16:
I used to use 1.5.3 B4. I just found out that you have updated version 1.6.2 B3. But when I downloaded the 1.6.2 B3 file, I didn't find the hex file? Why is that? I've looked for it in detail, but I still can't find it. I hope to get your help. 1.62 looks great. I'd love to be able to use it in the cloud right away.

vostrenkov, 2020-10-16:
https://github.com/FreeJoy-Team/FreeJoy/releases/tag/v1.6.2b3

uwlftoe17, 2020-10-16:
> 您需要从此处获取.zip文件：[https](https://github.com/FreeJoy-Team/FreeJoy/releases/tag/v1.6.2b3)：
> [//github.com/FreeJoy-Team/FreeJoy/releases/tag/v1.6.2b3](https://github.com/FreeJoy-Team/FreeJoy/releases/tag/v1.6.2b3)
> 
> 您还需要使用新的配置器。可以在配置器存储库的发行页面上找到它：[https](https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases/tag/v1.6.2b3) :
> [//github.com/FreeJoy-Team/FreeJoyConfigurator/releases/tag/v1.6.2b3](https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases/tag/v1.6.2b3)

Thank you Vostrenkov, but it seems so difficult. I still can't find what I need. I'm just an ordinary DIY fan. It's difficult for me to complete the installation of 1.6.23b. It would be great to have a complete teaching, but that's just my hope. I hope someone can help me.

vostrenkov, 2020-10-16:
You need to get .hex file from the .zip archive and burn it into your bluepill just like described in [wiki](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Flashing-firmware.md)

Then take archive with configurator utility from .zip archive on configurator release page and run .exe to start configurator

uwlftoe17, 2020-10-17:
> 您需要从.zip存档中获取.hex文件，并将其刻录到您的bluepill中，就像[Wiki中所述](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Flashing-firmware.md)
> 
> 然后从配置器发布页面上的.zip存档中获取配置器实用程序，并运行.exe以启动配置器

 I downloaded all the 1.6.2b0-b3.zip files, but I still couldn't find the 1.6.2b3.hex file. If you have this 1.6.2b3.hex firmware, I hope you can email me. Thank you very much. I'm looking forward to the new version FREEJOY.email:303379 @ qq.com   
Thank you Vostrenkov

uwlftoe17, 2020-10-17:
> You need to get .hex file from the .zip archive and burn it into your bluepill just like described in [wiki](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Flashing-firmware.md)
> 
> Then take configurator utility from .zip archive on configurator release page and run .exe to start configurator

I downloaded all the 1.6.2b0-b3.zip files, but I still couldn't find the 1.6.2b3.hex file. If you have this 1.6.2b3.hex firmware, I hope you can email me. Thank you very much. I'm looking forward to the new version FREEJOY.     email:303379 @ qq.com
Thank you Vostrenkov

vostrenkov, 2020-10-17:
Here is the direct link to the archive on releases page: https://github.com/FreeJoy-Team/FreeJoy/releases/download/v1.6.2b3/FreeJoy_v1_6_2b3.zip

[image]

uwlftoe17, 2020-10-18:
I got 1.6.3b.hex file. It's very powerful. I'm so happy. I'm surprised. Thank you very much

vostrenkov, 2020-10-18:
1.6.2b4 is already available

uwlftoe17, 2020-10-18:
I downloaded freejoy_ V1_ 6_ 2b4.zip. There are only three files in it. I can't find freejoy Configurator.exe program.Can you send me. Exe alone?

vostrenkov, 2020-10-18:
Configurator placed in its own repo. Yiu can download latest release here:
https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases/tag/v1.6.2b3

closed 2020-11-18 by vostrenkov