FreeJoy-Team/FreeJoy #64: addition of the AS5048A sensor

gerh, 2020-10-22:
I have added the as5048a rotation sensor to the FreeJoy and FreeJoycodeConfigurator code and placed it on my fork. Any interest in merging this into the main code?

vostrenkov, 2020-10-22:
Hey, 
Sure. You can make a PR and I will take a look

dermobb, 2020-11-04:
Oh I would love to see this chip added. It seems to be much easier to find in my area than for example the TLE5010/11. Thanks in advance!

vostrenkov, 2020-11-05:
I will be done but it can take a while because I can only add it to the next version to keep config compatibility

dermobb, 2020-11-05:
> I have added the as5048a rotation sensor to the FreeJoy and FreeJoycodeConfigurator code and placed it on my fork. Any interest in merging this into the main code? And there was a small glitch in the spi.c source that I removed. That gave extra pulses on the SPI_SCK line when using the as5048a.

That's great. Thanks very much. How can I use your code. Do I have to compile it?

dermobb, 2020-11-05:
> I will be done but it can take a while because I can only add it to the next version to keep config compatibility

Thanks for considering. In the meantime, how can I use the enhanced code of @gerh?

gerh, 2020-11-05:
It is better to wait using my code. (although you can download the source code, freejoy and freejoy configurator, from my github and compile it yourself) . I have run into a problem with my code: if only the AS5048A is attached, then everything works fine, but if I add a TLE5011 or other device things do not work. The first device in time after the AS5048A does not read the data. If I change the SPI_mode of the AS5048A from 1 -> 0, all the other devices are working well, but then the AS5048A will give the wrong data. So something goes wrong with changing the SPI phase between devices. I'm working on this...

dermobb, 2020-11-05:
> It is better to wait using my code. (although you can download the source code, freejoy and freejoy configurator, from my github and compile it yourself) . I have run into a problem with my code: if [...]

Thanks for your reply. In my current project (joystick) I would need two as5048a and a handful of buttons. I would happily try your version even knowing there are certain bugs. Only thing is, I don't know how to compile either firmware nor software. Can you provide a compiled version of both? What tools are needed to compile the code? Thanks

gerh, 2020-11-09:
I have placed a compiled version of both freejoy and freejoyconfigurator in the directory binhex in my gerh/FreeJoy repository. You can have a go at it. It should work with one AS5048A and some buttons, and probably with more AS5048A's. I can not check it, I have only one AS5048A. Please let me know. At the moment the AS5048A still does not work with other types of SPI devices.

gerh, 2020-11-09:
working version now

dermobb, 2020-11-09:
Thank you very much! I'm still in the planning phase and 3d printing parts but it is very good to have this version ready.

vostrenkov, 2021-01-10:
@gerh please check out the version placed in /dev brach. it was merged with your PR but some SPI code has been changed. I dont have AS5048A in stock so that would be cool if you could check it

gerh, 2021-01-30:
@vostrenkov I just checked the AS5048A with the newest code in the /dev branch and it is not functioning. I will have a look at it in the coming days.

gerh, 2021-02-03:
The code in my dev branch is working now, but only for one AS5048A with -O3 optimization in C, and it is working fine for more AS sensors when I turn the optimization off -O0. I tested the code with 2 AS5048A's and 2 TLE5011's.

gerh, 2021-02-05:
The code in my dev branch is working now for my 2x AS5048A's and 2x TLE5011's with every kind of optimization.

vostrenkov, 2021-02-07:
Okay, thanks, I'll take a look soon

vostrenkov, 2021-05-17:
Does it work as expected in the last version of released software? Can I close the issue?

gerh, 2021-05-18:
I checked version FreeJoy_1_7_0b3 with my two TLE5011's and two AS5048A's. And all things are working fine. So, yes, the issue can be closed.

closed 2021-05-18 by gerh