FreeJoy-Team/FreeJoy #66: D-Pad problem

Safandula86, 2020-10-24:
Hi, i try to use multiswitch ALPS RKJXT1F4200, the problem is that after configuring everything the D-Pad works fine except "D-Pad Up". I don't know why when I release from the Up position, the "D-Pad Center" lights up after a while. I have no problem with other directions, I checked it on 2 multiswitches and 2 boards.
https://youtu.be/LSdQ5xFOoj0
My configuration :
[image]
[image]

[Button Box.zip](https://github.com/FreeJoy-Team/FreeJoy/files/5433527/Button.Box.zip)

vostrenkov, 2020-10-27:
Hi!
I made some test and I can guess it is because of internal structure of this type of D-Pad. Electrically when you press any direction your Center pin (Push Com) is also triggered:

[image]

FreeJoy hanlde this by software excluding Center press when any of directions pressed. But there may be some kind of mechanical delay between releasing Up pin and Push pin, so FreeJoy can catch the moment when only Push pin is pressed and implement it as Center button press.

The way how you can handle this is to increase Button debounce delay, values of 100-150 should be enough:

[image]

Safandula86, 2020-10-27:
This only happens with Up. I also noticed that when we quickly press and release it is ok, but if we hold and release, Push will light up after a while.

ethan54013, 2021-07-10:
@vostrenkov Hi, I just ran into the same problem too, I have tried setting the debounce delay to 100-150 like you suggested, but the problem is still there. Only if you quickly press and release the Up direction, the problem will disappear, but it is very difficult to control. So I want to ask that is there any alternative solution to this? Thank you!

Mawasto, 2024-10-10:
Hi, i have the same issue, does anyone have a fix for that?