FreeJoy-Team/FreeJoy #68: Feature request

armor66, 2020-10-30:
This is a request in case it is possible to add. I gratefully use freejoy for open.hd(wifibroadcast), and would like to  use it wireless with RC Flight Simulator. Is it possible to add some serial protocol like SBUS to TXD PA9 or just hint where to start.

vostrenkov, 2020-10-31:
Yes, it is possible (but a bit not the goal of the project).
There is a place in code where HID report is sent to the host device. You can add serial output just there.
File stm32f10x_it.c:
```
if (ticks - joy_ticks >= dev_config.exchange_period_ms )		{			joy_ticks = ticks;							// getting fresh data to joystick report buffer			ButtonsGet(physical_buttons_data, joy_report.button_data, &joy_report.shift_button_data);			AnalogGet(joy_report.axis_data, NULL, joy_report.raw_axis_data);				POVsGet(joy_report.pov_data);						joy_report.raw_button_data[0] = btn_num;			for (uint8_t i=0; i<64; i++)				{				joy_report.raw_button_data[1 + ((i & 0xF8)>>3)] &= ~(1 << (i & 0x07));				joy_report.raw_button_data[1 + ((i & 0xF8)>>3)] |= physical_buttons_data[btn_num+i] << (i & 0x07);			}			btn_num += 64;			btn_num = btn_num & 0x7F;						joy_report.id = REPORT_ID_JOY;											USB_CUSTOM_HID_SendReport((uint8_t *)&joy_report.id, sizeof(joy_report) - sizeof(joy_report.dummy));		}
```

armor66, 2020-11-01:
Thank you so much. That's enough.

closed 2020-11-01 by armor66

Sergey-S-FF, 2024-07-12:
@armor66 Did you succeed after all?