FreeJoy-Team/FreeJoy #69: there's a stagnation zone at -9 ??

tinglengyue, 2020-11-09:
I used two kinds of op but They are the same in that respect, I think So Hardware problems almost can be ruled out . When the pressure bar goes to -9, the pressure increases a little bit and there's a stagnation zone at -9  Close to 1/2 of the voltage   but why ??

vostrenkov, 2020-11-09:
Please pin your configuration file

tinglengyue, 2020-11-09:
After i download bootloader  and app.i do nothing  . Just default config file

tinglengyue, 2020-11-18:
stm32F103C8T6    ADC at 0x400  0xc00  There's a nonlinear point here   ？？STM32 ADC bug？？？？The test uses stm32 small blue plate and a standard voltage source   At the same time use voltage source input voltage to A0 A1 A2  to test.

tinglengyue, 2020-11-18:
[image]
[image]

vostrenkov, 2020-11-18:
I cannot reproduce your issue. Can you confirm that it is repeated on different boards?

vostrenkov, 2020-11-18:
[image]

tinglengyue, 2020-11-18:
Yes, I can confirm that it is repeated on different boards。but  try debug with jtag 
i find adc values near 0x400  and 0xc00  did not move  when pressure increase a little bit.So, what I want to ask is there a bug in stm32f103 ADC?  Have you ever met this problem?

tinglengyue, 2020-11-18:
It's like you haven't met。This is a little awkward

tinglengyue, 2020-11-18:
This is a strange thing   Would you like to share  you sch?

vostrenkov, 2020-11-18:
I have ordinary bluepill board without any external components. I also test it on my freejoy [controller board](https://github.com/FreeJoy-Team/Controller-Lite) with same result (watching values with SWD debug monitoring).
I've never faced such problem, any chance you use fake bluepills with fake STM32 MCU?
Do you check ADC behaviour with freejoy code only or you also try with clean ADC project?

closed 2021-05-17 by vostrenkov