FreeJoy-Team/FreeJoy #72: 74HC165 Shift Register

0xfgarcia, 2020-12-01:
Hi,

Is this connection correct for a shift register to work?

A5 -> Latch
A4 -> Data
B3 -> SCK

74HC165 push buttons pins are disconnected. 
74HC165 Pull Down.

Thanks!

[image]
[image]
[image]

[image]

vostrenkov, 2020-12-02:
Hi!
Connections looks right but you need to add pull-up or pull-down resistors on shift register inputs to prevent this flickering

0xfgarcia, 2020-12-02:
Ok Yuri, thanks!

closed 2020-12-02 by 0xfgarcia