FreeJoy-Team/FreeJoy #73: Shift Modifiers on Encoders

AndyPeacock, 2020-12-01:
I have one encoder set up with logical outputs for unmodified (3,4), shift1 (5,6) and shift 2 (7,8) states. Shift1 (1) and shift2 (2) are controlled by a toggle switch and work correctly. When no modifier is active, logical 3 and 4 come on as I rotate the encoder. If I enable shift1, logical 5 and 6 come on as I rotate, as expected, but I also see 3 and 4 (unmodified state) come on too, the same applied for shift2. 

I understand that the unmodified logical output should not come on, if there is also a shifted output setup, but the shift is not active. _If physical button has both logical buttons with shift modificator and without them, then the logical button which has no modificator will work only if shift modificators are not pressed._ I'm on the device firmware 1.6.2b4.

[image]

vostrenkov, 2020-12-02:
Hi,
there's known issue on shifts with encoders. It will be fixed soon

vostrenkov, 2021-01-10:
should be fixed now

closed 2021-05-17 by vostrenkov