FreeJoy-Team/FreeJoy #75: 7 way encoder

vorph666, 2020-12-05:
Hi,
I´m building a steering wheel button box. I want to use two 7 way encoders. 
Up, down, left, right, turn left, turn right, press button. 

When now switching to any direction, the press button is also activated. Do you have any idea, how to fix this behavior?
It is ok when rotating left or right.

Thanks,
Thomas

[image]

snubbsyuk, 2020-12-06:
Hey,
You need to use the POV_center "function" for the push button - this will eliminate the issue you're facing.

Cheers

vorph666, 2020-12-07:
Great, that was easy. It works. Thanks!

closed 2020-12-30 by vostrenkov