FreeJoy-Team/FreeJoy #78: CD4021 pin mapping

geologic, 2020-12-23:
Hi

All the examples i saw here with shift registers uses 74HC165, my local supplier only has CD4021, but they are not pin equivalent.
Can you please help me with the wires? I did like this:

pin 3 - shiftreg_DATA
pin 10 - SPI_SCK
pin 11 - shiftreg_LATCH
pin 8 - GND
pin 16 - 3V3
pushbutton on pin 1 connected to GND, with 10K pullup

i set CD4021_Pullup with 1 button count, but when i press the button nothing happens on Button Config tab.

Thanks

tested

vostrenkov, 2020-12-24:
Hi,
pin 11 is serial input for daisy chaining the registers. You should connect LATCH to pin 9 (Parallel/Serial Control)

[image]

In case of single register Pin 11 (Serial In) may be left unconnected or tied to GND/VCC. In case of multiple registers you should daisy chaing them by connecting Pin 11 (Serial In) to next register's Pin 3 (Q8). LATCH is shared between all registers in chain.

geologic, 2020-12-29:
Great, thanks!

closed 2020-12-29 by geologic