FreeJoy-Team/FreeJoy #82: Adding support for the 16 bit shift register from suncom F15 talon realtek RTS752B

drepou, 2021-01-05:
Hi Nice works guys ! work great

Can you Add 16 bits shift register implementation for F15 talon ? 

the shift register used is 16 bits realtek RTS752B from what i read here but i dint found the datasheet

http://theseger.com/projects/2014/04/converting-old-joysticks/

They are connected to a Parallel to Serial chip inside the grip. To query button data, you need to clock it out of the chip. you control Pi and CS and receive data from D0-D1.
start with Pi and CS at 0. set Pi to 1, then CS to 1, then Pi to 0 then CS to 0.
Then repeat the following 16 times to get all 16 buttons:
set cs to 1, read D0 (1=button pressed 0=released)
then set cs to 0.

For more details, take a look at the PollButtons(void) function in the source code.

static void PollButtons(void)
{
	static BYTE count = 0;
	BYTE i;
	GBSTART = 1;
	buttLow = 0;
	GBCLK = 1;
	buttHigh = 0;
	GBSTART = 0;
	buttLow = 0;
	GBCLK = 0;
	for (i=0; i<16; i++)
	{
		GBCLK = 1;
		buttHigh >>= 1;
		if (GBDAT == 0)
			buttHigh |= 0x80;
		GBCLK = 0;
		if (i==7)
			buttLow = buttHigh;		
	}
	// remap some buttons
	i = buttLow;
	buttLow = 0;
	if (i & 1)	buttLow |= 0x20;
	if (i & 2)	buttLow |= 0x80;
	if (i & 4)	buttLow |= 4;
	if (i & 8) buttLow |= 1;
	if (buttHigh & 0x1) buttLow |= 0x40;
	if (buttHigh & 0x2) buttLow |= 0x10;
	if (buttHigh & 0x4) buttLow |= 8;
	if (buttHigh & 0x8) buttLow |= 2;
	buttPov = hatpos[i >> 4];

	// handle base buttons
	if (count == 0) {
		if (oldswmain != swmain) {
			oldbutts = baseButtons - 1;
			oldswmain = swmain;
		}
		if (((oldbutts ^ baseButtons) & 0x0F) != 0) {
			if (swmain == 0)
				curbutts = (PORTB ^ 0xFF) & 0xF;
			else {
				curbutts = 0;
				TEST_BUTTON(1, mLED_1);
				TEST_BUTTON(2, mLED_2);
				TEST_BUTTON(4, mLED_3);
				TEST_BUTTON(8, mLED_4);
			}
			oldbutts = baseButtons;
			count = 50;
		}
	} else {
		count--;
	}

}

thanks !

vostrenkov, 2021-01-07:
Hi!
looks like it is very uncommon solution and this chip isnt easy to be found in stock. So there is very small chance that support will be added

drepou, 2021-01-07:
OK thanks for answer i know that this old 16 bits shifter is not common but many people are converting these old suncomm F15 and it can avoid to create a matrice of button and soldering by juste keeping the shifter . i will try to edit the code myself if possible ( i think it not very difficult since it a quite easy process similar to the one implemented already just need to do it 16 times instead of 8 ... ) i didnt watch if the code is open source . i will tell you and offer a pull resquest if it work ( not shure it will accept 3.3v as VCC ) will see or use a more common shifter but the one implemented can drive only 8 button so i need 2 of them....

closed 2021-01-07 by drepou