FreeJoy-Team/FreeJoy #84: Deadband

0xfgarcia, 2021-01-12:
I am testing deadband option as shown in the documentation, I can't get it to work, adjusting deadband the output bar of the selected axis goes to jumps, it does not leave the dead zone fixed at the same level whenever the axis, and sometimes when you lower the axis to 0 (-32767) it leaves the output activated to the minimum.

It also allows adjusting from 0 to 127 which I do not quite understand and the documentation does not clarify the reason for that figure or the effect it has. 

Thanks!

Freejoy 1.6.2b4
Qt Configurator 1.6.2 - 1.1.2

[image]

[FreeJoy.cfg.txt](https://github.com/FreeJoy-Team/FreeJoy/files/5801297/FreeJoy.cfg.txt)

vostrenkov, 2021-01-12:
Hi,
dynamic deadband algorithm has some math to detect movement of the axis and hold the output value when movement is too "week". Deadband digits determine the threshold of detecting - higher values is more agressive deadband and you will have to move the axis faster to make the value change

0xfgarcia, 2021-01-12:
Thanks for the quick reply, I understand now.

Is there a parameter in the configurator to eliminate the dead zone at the beginning of the analog axis? I am working with a load cell and I need to eliminate a dead zone at the beginning of the axis between -32,767 to -32,500

Thank you!

vostrenkov, 2021-01-12:
Yes, you can just manually tune values in calibration section. Decrease Maximum a bit and increase Minimum to get deadzones on start and and of the axes

closed 2021-01-13 by 0xfgarcia