FreeJoy-Team/FreeJoy #85: Need mlx 90316  support

hde610, 2021-01-12:
I know freejoy can  support 90393,but i have to use 90316 ,Are they compatible？

vostrenkov, 2021-01-13:
No, these sensors are not compatible

virtual812, 2021-06-13:
Just putting this here.
Not really making a request, but if you guys wanted to add support for the Thrustmaster Warthog sensor it's a MLX90333
Probably unlikely anyone would separate it from it's existing controller, also unlikely anyone would use the same sensor in a DIY solution.
That said if you guys got bored (unlikely?) and it's implementation is similar to other sensors in the family you might consider implementing it?

vostrenkov, 2021-06-16:
Best way to add this sensor to freejoy is to make a PR with its implementation (just like it was done with MLX90363) ;)
Time for working on the project is very limited now and adding new sensor require having it in stock to do some testing

drepou, 2021-09-01:
I get a defective trustmaster wharthog base for 15$ it would be so cool to support for the Thrustmaster Warthog sensor MLX90333 on stm32... but may be the way to make my base work is to replace the original sensor with the sensor 90393 and use my stm32 board on it ....