FreeJoy-Team/FreeJoy #86: MLX90393 SPI

SarwarN88, 2021-01-20:
Hi vostrenkov,
I am trying to communicate with the MLX90393 from Arduino SPI library and facing issues to call start single measurement SM and read measurement RM command. I get values but they are all jumping and its not reading magnetic strength. I been working with the code for more than 2 weeks with no results. I don't know where the problem is. I am following what the data sheet is telling me but the sensor is not responding to request SM and RM commands. I know you already worked on the sensor before and added to your library using SPI mode. I was wondering if you could help me making the sensor read measurement using SPI library from Arduino. I know its a big favour to ask but you are my only hope. I been searching everywhere for SPI code but all is for I2C. I can make the code work with I2C mode but its SPI mode that is not responding. And yes I followed the wiring diagram from the data sheet cutting the jumper line CS on the board and connected to the ground to activate SPI mode. The sensor is sending data as far as I know but not in the way I am asking for.
Bellow is my code:
```
#include<SPI.h>
const uint8_t CS = 7; // slave select pin
SPISettings settings(2000000, MSBFIRST, SPI_MODE3); //2 MHz clock, MSB first, mode 3

void setup() {
digitalWrite(CS, HIGH);  // set the SS pin HIGH
SPI.begin();
// Initialise serial communication, set baud rate = 9600
Serial.begin(9600);
delay(300);
}

void loop() {
unsigned int data[7];
SPI.beginTransaction (settings);  // 2 MHz clock, MSB first, mode 3
digitalWrite (CS, LOW); // set the SS pin LOW
delayMicroseconds(1);
// Start single meaurement mode, ZYX enabled
SPI.transfer(0x3E);
digitalWrite(CS, HIGH);
SPI.endTransaction();
delay(100);

SPI.beginTransaction (settings);  // 2 MHz clock, MSB first, mode 3
digitalWrite(CS, LOW);
delayMicroseconds(1);
// Send read measurement command, ZYX enabled
SPI.transfer(0x4E);
digitalWrite(CS, HIGH);
SPI.endTransaction();
// Request 7 bytes of data
SPI.beginTransaction (settings);  // 2 MHz clock, MSB first, mode 3
digitalWrite(CS, LOW);
delayMicroseconds(1);
// note: use dummy data 0 or 0x00 to force spi bus to send data (maybe 7 dummy data)
// Read 7 bytes of data
// status, xMag msb, xMag lsb, yMag msb, yMag lsb, zMag msb, zMag lsb
data[0] = SPI.transfer(0);
data[1] = SPI.transfer(0);
data[2] = SPI.transfer(0);
data[3] = SPI.transfer(0);
data[4] = SPI.transfer(0);
data[5] = SPI.transfer(0);
data[6] = SPI.transfer(0);
digitalWrite(CS, HIGH);
SPI.endTransaction(); 

// Convert the data
int xMag = data[1] * 256 + data[2];
int yMag = data[3] * 256 + data[4];
int zMag = data[5] * 256 + data[6];
 
// Output data to serial monitor
Serial.print("Magnetic Field in X-Axis : ");
Serial.println(xMag);
Serial.print("Magnetic Field in Y-Axis : ");
Serial.println(yMag);
Serial.print("Magnetic Field in Z-Axis : ");
Serial.println(zMag);
delay(50);
}
```
This what I am reading. The values for X,Y,Z are jumping and not responding to magnet movement.
[image]
 
My wiring is done as the diagram showing CS jumper line cut and connected to the ground to read 0 for SPI mode. Also used 4 wiring mode MOSI MISO are not shorted.
[image]

vostrenkov, 2021-01-20:
Hey!
First if all, do you have self-made pcb or you use one from aliexpress or amazon?
If the 2nd please check MOSI and MISO lines marking on your board (check shorts to the chip pins). Some of these boards have wrong marking as i mentioned in [FreeJoy Wiki](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/MLX90393-connection.md).
Also you say you connected CS to the ground, does that means CS is **shorted** to the ground or pulled-down by a resistor? I use CS line in Push-Pull mode on my controller and only tie it to logic 0 while the transmission.
I never tried 3-wire SPI mode for MLX90393 so all my code checked with 4-wire connection. You can find the most actual code for communication with sensor in the /dev branch. I made some experiments with setting MOSI in OD mode, you may try this also

SarwarN88, 2021-01-21:
Thanks for the quick respond. I bought mine from AliExpress its CJMCU-90393 same like the one you have used in the project. I have two of them. It mentioned in the Adafruit website that MOSI and MISO need to be connected in opposite way to the master due to print error on the board. I can confirm that reading showing only when they are connected in a opposite way.  I shorted CS to ground directly. Following you advise I just pulled-down the CS pin with 10K resistor to ground but the problem is still the same. I will try to test my code on the the other sensor and maybe with Arduino Uno since I am doing the test with Maple Mini. I might try your library for the sensor if all failed. By using your library mlx90393.h and mlx90393.cpp how I can call function command to read measurement? Can you give me a simple example. Thanks

vostrenkov, 2021-01-22:
You dont even need CS pull down resistor. Use direct connection and push-pull mode for your contoller's pin.

My startup sequence:

1. Exit command
2. Delay 20ms
2. Reset command
3. Delay 50ms
4. Read register 0
5. Set GAIN=0 and HAL_CONF=0
4. Write register 0
5. Delay 20ms
6. Read register 1
7. Set bits SPI_MODE, BURST_SEL_X, BURST_SEL_Y, BURST_SEL_Z 
5. Write register 1
6. Delay 20ms
7. Read register 2
8. Set RES=1,1,1 OSR=0 and DIG_FILT=0
6. Write register 2
7. Delay 20ms
7. Enable burst mode
8. Delay 20ms

Then I read measurement data in a cycle. All this code can be found in mlx90393.c file. You can adapt it to your code by changing low level SPI and delay functions. And also read measurements in blocking mode since arduino dont have DMA

SarwarN88, 2021-01-26:
Thanks for the responds and your supports. I finally managed to make SPI mode work with Maple Mini using adafruit/Adafruit_MLX90393_Library. This library works with Arduino UNO in SPI mode but was failed uploading to Maple Mini or STM32. It was related to conversion between const uint8_t and uint8_t. I fixed that error by changing `size_t write` to `const uint8_t` in WireBase.h and WireBase.cpp in STM32 library in Wire folder. I hope those changes won't create conflict in other libraries linked to the Wirebase.

closed 2021-01-26 by SarwarN88